lgsvl/simulator #529: About OPENDRIVE and APOLLO5MAP

yumianhuli2, 2019-11-29:
Hello，Does the current map have slope information? Is it now supported to make rolling roads based on maps?Thank you！

martins-mozeiko, 2019-12-02:
OpenDrive supports height information in hdmap. In current simulator version Apollo does not export height data.

ntutangyun, 2020-01-09:
@martins-mozeiko since the format actually supports height information, does it mean that we could build the map ourselves and manually add in the height information for Apollo to consume ? thanks

luqiang21, 2020-01-09:
@ntutangyun Yes, you can add height information yourself to the map by modifying map exporting tools in our simulator. It depends on whether Apollo will parse height information from the map or not.

ntutangyun, 2020-01-10:
@luqiang21 alright. thank your for your prompt reply.

closed 2020-04-08 by EricBoiseLGSVL