lgsvl/simulator #530: IMU and GPS Message delay

clouddd9, 2019-12-02:
When I make the co-simulation, the apollo Dreamview shows messages as follows

Raw IMU Message Delay. IMU message is 178 cycle 1.78812 sec behind current time.09:52:11
Raw GPS Message Delay. GPS message is 178 cycle 1.78797 sec behind current time.09:52:11
Raw IMU Message Delay. IMU message is 178 cycle 1.78818 sec behind current time.09:52:11
Raw GPS Message Delay. GPS message is 179 cycle 1.79801 sec behind current time.09:52:11
Raw IMU Message Delay. IMU message is 178 cycle 1.78838 sec behind current time.09:52:11
[image]

And I locate the problem in the rtk_localization.cc from apollo-5.0/modules/localization/rtk/
[image]

What's wrong with the GPS and IMU timestamps? How to solve it? Could you give me some advice? :)

yumianhuli2, 2019-12-02:
> When I make the co-simulation, the apollo Dreamview shows messages as follows
>
> Raw IMU Message Delay. IMU message is 178 cycle 1.78812 sec behind current time.09:52:11
> Raw GPS Message Delay. GPS [...]

Perhaps because the imu is updated too infrequently

martins-mozeiko, 2019-12-02:
Is something not working? We see these kind of messages from time to time, but everything else works.

Martin20, 2020-02-04:
Could you solve this error?
In front of GPS in the module controller, it is written FATAL ERROR.

How can I fix those issues?

askariz, 2020-04-16:
I have also come across this problem . 
enviroment :
simulator pc :  run simulator 192.168.1.114
apollo pc : run apollo code stack .192.168.1.117
[image]

373432156, 2020-04-29:
same issue！！

rongguodong, 2020-09-25:
This is because our timestamp is simulation time, not real time.
We have added sim_time support to latest Apollo. If you add a ClockSensor to ego, and set CyberRT's `clock_mode` to `MODE_MOCK`, you should not see these errors. For more details, please refer to [our document about ClockSensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#clock).

closed 2020-10-06 by EricBoiseLGSVL