lgsvl/simulator #532: import hd map

yumianhuli2, 2019-12-02:
I downloaded the CrossingComplex8Course map from the opendrive website and clicked the import hd map button, but nothing came back. Is this function valid only if you enter the simulator state?

martins-mozeiko, 2019-12-02:
I've tried importing this map and it worked fine.
Make sure you turn on annotations in "Simulator -> Annotate HD Map" menu, click "View All" button to see all the annotations. Note that they might be behind, above or below you. You'll need to rotate camera, or focus on annotations in scene view.

yumianhuli2, 2019-12-03:
> I've tried importing this map and it worked fine.
> Make sure you turn on annotations in "Simulator -> Annotate HD Map" menu, click "View All" button to see all the annotations. Note that they might be behind, above or below you. You'll need to rotate camera, or focus on annotations in scene view.

Yes,it works.But the scene view is black in unity2019.1.10f1, other version supports?
Have you ever been in this situation？

martins-mozeiko, 2019-12-03:
Can you show screenshot - what exactly is black?

If you are working on creating map, then 3d models will be dark because there are not lights in scene. You need to turn off "Scene lighting" - lightbulb icon in Scene view. Or temporarily add Sun prefab to scene hierarchy.

yumianhuli2, 2019-12-04:
> lightbulb icon

The windows version is fine,I can see the map and light is on.
But In linux version I mean black is like below:
[image]

martins-mozeiko, 2019-12-04:
This map does not seem to have any 3d objects in scene hierarchy. Only annotations. To see annotations you need to click "View All" button in "Simulator -> Annotate HD Map" tool window.

yumianhuli2, 2019-12-04:
> This map does not seem to have any 3d objects in scene hierarchy. Only annotations. To see annotations you need to click "View All" button in "Simulator -> Annotate HD Map" tool window.

I tried,And I created cube/sphere int the Hierarchy,still can not see anything in this project!But another project is fine.
[image]

martins-mozeiko, 2019-12-04:
The annotations are not always centered around origin of game object. You might need to look behind you or above you. Select one specific lane or lane segment and double click it, to focus on that. Make sure console does not have any errors after import.

From screenshot it seems that you have two map annotation holders - `MapHolder`, and `MapCrossingComplex8Course`. That may lead to some issues. Delete one that is not used.

Also in this screenshot I do not see that you have added any sphere or cube in hierarchy. Simply clicking right click and choosing "3D Object -> Plane" should be enough for ground. As long as you position it correctly car will be able to drive on it.

yumianhuli2, 2019-12-05:
> The annotations are not always centered around origin of game object. You might need to look behind you or above you. Select one specific lane or lane segment and double click it, to focus on that. [...]

[image]
Hello! I did everything you said, but I still can't see anything. But I do the same thing on Windows that is fine.

martins-mozeiko, 2019-12-05:
Can you check if you have map annotation Gizmos enabled? The Gizmos button should be also pressed.
Map annotations are drawn in Editor as Gizmos which can be hidden.

[image]

yumianhuli2, 2019-12-05:
> Can you check if you have map annotation Gizmos enabled? The Gizmos button should be also pressed.
> Map annotations are drawn in Editor as Gizmos which can be hidden.
> 
> [image]

Thank you for your reply！
[image]
I have pressed the map annotation！

EricBoiseLGSVL, 2020-04-08:
@yumianhuli2 You are in ortho view mode, I suggest perspective.  Select an annotation object in the Hierarchy panel and press 'F' to focus on that object in Scene view.  Last thing, look at annotation objects transforms in Inspector panel.  If there are floating point errors messages and the values are extremely high, then you have issues importing the HDMap and MapOrigin settings.

closed 2020-04-08 by EricBoiseLGSVL