lgsvl/simulator #534: Use the simulator with ROS without using Autoware or Apollo

Felixlxc, 2019-12-02:
Hi, 
    I am working on using the simulator with ROS and controlling the ego vehicle through ROS routine. I wonder if I can control the vehicle without using Autoware or Apollo so that I can use my own ros package, for example, publish the "/vehicle_cmd" to control the ego vehicle in simulator throught ROS bridge.

martins-mozeiko, 2019-12-02:
Yes, it is possible to control vehicle in simulator without full Apollo or Autoware running.

You can send exactly same message type as Autoware or Apollo is using to simulator, which will control the vehicle.

If you would want to use different message type, that would require modifying code a bit to accept new message type.

Felixlxc, 2019-12-02:
@martins-mozeiko  Thanks for your quick reponse. Then for controlling the ego vehicle, is it the rostopic "vehicle_cmd" of message type geometry_msgs being used? Or is there a reference document for how to design the rostopic to control the ego vehicle in simulator?

martins-mozeiko, 2019-12-02:
In case of Autoware its /vehicle_cmd topic expects `autoware_msgs/VehicleCmd` type. You can get it from Autoware repository.

In case of Apollo 3.0 control message has `pb_msgs/ControlCommand` type.

We don't really have any document showing how to send this message. But regular ros message publishing would work just fine. You should check the source to see which exactly and how the message fields are used: [Assets/Scripts/Bridge/Ros/RosConversions.cs#L425](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosConversions.cs#L425)

Felixlxc, 2019-12-02:
@martins-mozeiko Thanks for your help!

closed 2019-12-02 by Felixlxc

LawranceXcl, 2019-12-05:
Hi,Felixlxc. I have a question about lgsvl simulator that  the simulator itself hasn't control model, so when  I just will validate the algorithm of action plan like lane change, it must be connected  with autoware or appollo to use their control modules, isn't it? Do you known Carla Simulator? It itself has simple vehicle controls methods.

Felixlxc, 2019-12-05:
@LawranceXcl  Hi, Just like @martins-mozeiko  said, you can control the vehicle without using Autoware or Apollo but you need to publish the same ros topic with what Autoware or Apollo publish for controlling the vehicle.

LawranceXcl, 2019-12-05:
Thks! Felixlxc.