lgsvl/simulator #536: Error after building a bundle,Error while creating a bundle

weiyinxie, 2019-12-03:
hey guys i have an error after i press build assetsbundles and i got this message

[image]

i have installed 
ubuntu 16.04 
Unity Hub v2.0.0 Release
Unity 2019.1.10f1
Nodejs 12.13.1 lts 
lgsvl simulator 2019.11-rc1
on my system. so it's said that the incompability between the simulator and the linux operating system in the old release has already been fixed in the latest version, so how could this error happen? is there any solution to fix this issue? Thx a lot!!!

martins-mozeiko, 2019-12-03:
One error message above explains that Windows support is not installed in your Unity. When we build asset bundles we build it for both - Windows and Linux, because we want to be able to load same file on both OS'es.

To fix this, run Unity installer and add Windows target support. After that bundle building should work.

weiyinxie, 2019-12-04:
@martins-mozeiko 
Thx a lot i have reinstalled the unity hub with adding the Windows target support into the newly installed unity hub. The old issue has gone while new errors and warning occur.....do u probably know why?? Thx!
[image]
[image]

martins-mozeiko, 2019-12-04:
These are not real errors. Unity likes to complain about some internal warnings/errors. But the build should still succeed. Check if output files are produced.

weiyinxie, 2019-12-05:
@martins-mozeiko the output sucess....thx alot! have a nice day!!

closed 2019-12-05 by weiyinxie