lgsvl/simulator #537: opendrive import error

yumianhuli2, 2019-12-03:
[image]
I choosed the map below from opendrive official website：
[image]

martins-mozeiko, 2019-12-03:
Are you sure you are selecting same file when importing it on Linux?

I have tried it, and it fails with same error on Linux for me. This is because this map is created with OpenDrive 1.1. There are differences between different versions of OpenDrive. We currently support only 1.4 version.

If you would replace all instances of `level= "0"` in Crossing8Course.xodr file to `level= "false"` then the map can be imported. They changed this attribute type at some version of OpenDrive.

yumianhuli2, 2019-12-04:
> Are you sure you are selecting same file when importing it on Linux?
> 
> I have tried it, and it fails with same error on Linux for me. This is because this map is created with OpenDrive 1.1. There are differences between different versions of OpenDrive. We currently support only 1.4 version. Older or newer versions may or may not work, depending on contents of file.
> 
> If you would replace all instances of `level= "0"` in Crossing8Course.xodr file to `level= "false"` then the map can be imported. They changed this attribute type at some version of OpenDrive.

ok solved it ! But 
1 Why not just generate the mesh ？
2 And then How do I use this map to integrate with other parts, and what are the steps？
Thank U!

martins-mozeiko, 2019-12-04:
Map import/export in our Simulator only deals with annotations. It does not create mesh as that can be very complex process depending on precision or quality requirements.

In simplest case you could create flat plane and use it as road.

closed 2020-04-08 by EricBoiseLGSVL