lgsvl/simulator #538: Feature discuss: Choose whether to enter api mode at startup?

daohu527, 2019-12-04:
Hi all,
I want to deploy the simulator automatically and then run the simulator through the python API. But now you have to start the api mode in the web ui before you can load the scene through the python api.  

I think it should be set to enter api mode instead of web ui at startup, otherwise large-scale automated deployment and configuration cannot be achieved.  

Is it possible to set in config.yml to let the simulator enter api mode directly? In this way, after the simulator is deployed, you can directly perform automated testing through the api.

martins-mozeiko, 2019-12-05:
Hi.

Currently you cannot do that with config.yml, but you can send same commands as HTTP request to simulator (port 8080) to start simulation in API mode. Web interface does all interactions (create/edit/delete map/vehicle/simulation) via simple json requests.  You can check what REST calls web interface is doing with debugger in browser, or simply look up what REST requests simulator is responding in Assets/Scripts/Web/Modules folder.

We're not yet sure how we will automate this for cloud deployment, but this is something we will work in next few months.

daohu527, 2019-12-07:
You're right, I need to deploy simulators on a large scale in the cloud, I'm going to try something, if there are good ideas, welcome to discuss. : )

EricBoiseLGSVL, 2020-04-08:
@daohu527 We are currently working on this feature and will be releasing news about it in the coming months

closed 2020-04-08 by EricBoiseLGSVL