lgsvl/simulator #542: Building asset bundles not working for vehicles [Ubuntu build]

MithunKinarullathil, 2019-12-05:
**Description:**
-  Jaguar2015XE not building as AssetBundle in Ubuntu

**How to reproduce:**
- Fresh clone the sim
- Download CubeTown and Jaguar2015XE to the Environments and Vehicles directories
- Build using the Editor tool without the Build Simulator checked
- Confirm that the AssetBundles directory does not contain Jaguar2015XE

**Refs:**

[image]

Errors are as shown in the figure.

MithunKinarullathil, 2019-12-05:
It's building after a few trials.

closed 2019-12-05 by MithunKinarullathil

martins-mozeiko, 2019-12-05:
The actual error is in second message from bottom - it says that you did not have Windows support installed for your Unity. When we build asset bundles for maps and vehicles we build it for two platforms - Linux and Windows, so result can be used on any supported OS.

MithunKinarullathil, 2019-12-06:
@martins-mozeiko  It would be a good idea to add this here, https://www.lgsvlsimulator.com/docs/build-instructions/. Is the documentation open for public contribution?

MithunKinarullathil, 2019-12-06:
@martins-mozeiko Thanks.