lgsvl/simulator #545: Where can I download apollo3.5 simulator

yumianhuli2, 2019-12-08:
Hello!
Where can I download apollo3.5 simulator？

martins-mozeiko, 2019-12-08:
There is no special version for Apollo 3.5. All available simulator versions support Apollo 3.0, 3.5 and 5.0 version. Just download latest code, it will work with Apollo 3.5.

yumianhuli2, 2019-12-08:
> There is no special version for Apollo 3.5. All available simulator versions support Apollo 3.0, 3.5 and 5.0 version. Just download latest code, it will work with Apollo 3.5.
Thank you！I mean where I can download the old simulator version .

martins-mozeiko, 2019-12-08:
Old simulator code is available in separate repository: https://github.com/lgsvl/simulator-2019.05-obsolete

yumianhuli2, 2019-12-08:
> Old simulator code is available in separate repository: https://github.com/lgsvl/simulator-2019.05-obsolete

very nice!

closed 2019-12-09 by martins-mozeiko