lgsvl/simulator #546: opendrive code generation

yumianhuli2, 2019-12-08:
Hello!I saw this from the source code.
//------------------------------------------------------------------------------
// <auto-generated>
//     This code was generated by a tool.
//     Runtime Version:4.0.30319.42000
//
//     Changes to this file may cause incorrect behavior and will be lost if
//     the code is regenerated.
// </auto-generated>
//------------------------------------------------------------------------------

// 
//This source code was auto-generated by MonoXSD
//

what about MonoXSD?where can I download this tool?
What exactly are the inputs and outputs of this tool?
Thank you！

martins-mozeiko, 2019-12-08:
This file is generated with xsd tool. xsd is available by default either as part of .NET SDK installation on Windows, or as part of [mono](https://www.mono-project.com/). Read documentation on MSDN website: [XML Schema Definition Tool](https://docs.microsoft.com/en-us/dotnet/standard/serialization/xml-schema-definition-tool-xsd-exe)

yumianhuli2, 2019-12-08:
[image]
So 
step 1 I should use XML to XSD
step 2 I should use XSD to classes?
Can you show me the command for converting the opendrive XML file to C# class. Thank U！

martins-mozeiko, 2019-12-09:
We use xsd to generate xml parser from xsd schema. OpenDrive provides xsd schema files here: http://www.opendrive.org/download.html

To generate xml parser code, run `xsd file.xsd /classes`

yumianhuli2, 2019-12-09:
> We use xsd to generate xml parser from xsd schema. OpenDrive provides xsd schema files here: http://www.opendrive.org/download.html
> 
> To generate xml parser code, run `xsd file.xsd /classes`

Hi！
1 You mean if I use  xsd OpenDRIVE_1.4H.xsd  /opendrive(class name),then I can get a opendrive class(a mapping file of opendrive xml)?
2 where can I download the Xsd.exe?May you provide me with the download address? Or installation steps？
Thank U very much！

martins-mozeiko, 2019-12-09:
1. Yes
2. If you are on Windows you need to install .NET SDK. On Linux you need to install mono. EIther of these options will provide `xsd` executable.

closed 2020-04-08 by EricBoiseLGSVL