lgsvl/simulator #547: How does vehicle dynamics support the simulation of any initial state?

yumianhuli2, 2019-12-08:
How does vehicle dynamics support the simulation of any initial state?
For example, let the vehicle dynamic simulation from the initial state of 80 km/h ？How does this work in your simulator？

martins-mozeiko, 2019-12-08:
You can set initial state if you use Python API. It allows to create EGO or NPC vehicle with any initial state, which includes both - position and velocity.

yumianhuli2, 2019-12-08:
> You can set initial state if you use Python API. It allows to create EGO or NPC vehicle with any initial state, which includes both - position and velocity.

https://github.com/lgsvl/simulator-2019.05-obsolete
This obsolete version supported Python API for setting vehicle initial state?And where is the doc for this python API?
But in my memory, this speed seems to have only get and no set method，How does the python API do that？

daohu527, 2019-12-09:
You can refer to [python API project](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/04-ego-drive-straight.py)  
Below is the code to set state of the ego car:  
```
state = lgsvl.AgentState()
state.transform = spawns[0]

forward = lgsvl.utils.transform_to_forward(spawns[0])

# Agents can be spawned with a velocity. Default is to spawn with 0 velocity
#  ------------------------------The speed of the car, you can change to 80--------------------------
state.velocity = 20 * forward            
a = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, state)
```

yumianhuli2, 2019-12-09:
> You can refer to [python API project](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/04-ego-drive-straight.py)
> Below is the code to set state of the ego car:
>
> ```
> state = [...]
oh，very nice!
python API project ialso supports https://github.com/lgsvl/simulator-2019.05-obsolete ？

martins-mozeiko, 2019-12-09:
Please read the documentation on supported features: https://github.com/lgsvl/simulator-2019.05-obsolete/tree/master/Docs/docs
Python API documentation is here: https://github.com/lgsvl/simulator-2019.05-obsolete/blob/master/Docs/docs/python-api.md

We do not provide support for old simulator anymore.

closed 2020-04-08 by EricBoiseLGSVL