lgsvl/simulator #549: cannot find the entry point of a c# script in unity to the simulation

weiyinxie, 2019-12-09:
Hello every one i'm pretty a new bird to Unity and trying now to do some simulation using the lgsvl which bases on the Unity. As i  cloned the  the simulator repository from page and opened the C# script of LidarSensor i cannot find the Main Func as Entry point of the Program. After doing some reasearch i found out that the entry point of the simulation is via the Unity engine and  for the common users MonoBehaviour is the only entry point to get our object in control. its essential to define the func start() and the func update() inside a c# script. But as i see the update() and start() is the private func and can only be accessed  via the func inside that class. On the other side the start() and update() func inside the c# script of LidarSensor have 0 reference, which means that no other public func have ever used them. So how could the both func interact with the unity engine.
[image]

martins-mozeiko, 2019-12-09:
MonoBehavior methods are called from Unity engine regardless of visibility. It calls them from internal native C++ code and finds methods via reflection by name. This way performance is better, because it avoids virtual function overhead.

I find diagram on [this page](https://docs.unity3d.com/Manual/ExecutionOrder.html) very useful to understand which methods of MonoBehavior are called by Unity and in which order.

weiyinxie, 2019-12-09:
@martins-mozeiko so it means as long as the member func of the object was inheritated from the monor behaviour, the unity enigine can load and excute them regardless of the visibility of them, even though they r declared as private member function.......am i correct???? amazing......

martins-mozeiko, 2019-12-09:
Yes, that is correct. Mono .NET engine knows everything about what methods class has. So it call it directly ignoring protection keywords.

weiyinxie, 2019-12-09:
@martins-mozeiko Thank u very much! Have a nice day!

closed 2020-04-08 by EricBoiseLGSVL