lgsvl/simulator #550: Route Editing Tab could not be selected And Bridge status is Null

kanbei, 2019-12-09:
Hello. I launched  Apollo and the Simulator according to the direction on website,
but in  Apollo, the Route Editing Tab could not be selected. 
In the Simulator, bridge status is Null.
I have input `bash scripts/bridge.sh`, like this photo, but I `m not sure whether the bridge is  launched successfully.
So how could I solve the problem.
[image]
[image]

OutOfCage, 2019-12-09:
Did you add the sensor configuration for your vehicle? 
[How to add vehicle](https://www.lgsvlsimulator.com/docs/vehicles-tab/)
And did you set the host and port of the bridge when adding a simulation?
[How to add simulation](https://www.lgsvlsimulator.com/docs/simulations-tab/#how-to-addedit-a-simulation)

kanbei, 2019-12-17:
Thank you. I have followed the instruction, and set senor configuration and port of the bridge, Apollo and the simulation was connected well. But now still have 3 problems.
The first one is if I set JSON Configuration of camera, according to the example(https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/), the simulation was crashed suddenly after I created the simulation and clicked 'play' button. 
When I deleted the configuration of camera, the simulation worked well. How could  I set the configuration of camera?

The second one is the perception and traffic light module can't be enabled in Apollo. I tried to click them but soon after it became unable. 

The third problem is when I added point and send routing request, the car didn't run and was still in the initial position. 
I think the position information in the simulation can be send to Apollo successfully because in Apollo,  the car could run well while I was operating the car manually in the simulation. But I don't know why car can't run automatically.
[image]

kanbei, 2019-12-17:
> Did you add the sensor configuration for your vehicle?
> How to add vehicle
> And did you set the host and port of the bridge when adding a simulation?
> How to add simulation

Thank you. I have followed the instruction, and set senor configuration and port of the bridge, Apollo and the simulation was connected well. But now still have 3 problems.

The first one is if I set JSON Configuration of camera, according to the example(https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/), the simulation was crashed suddenly after I created the simulation and clicked 'play' button.
When I deleted the configuration of camera, the simulation worked well. How could I set the configuration of camera?

The second one is the perception and traffic light module can't be enabled in Apollo. I tried to click them but soon after it became unable.

The third problem is when I added point and send routing request, the car didn't run and was still in the initial position.
I think the position information in the simulation can be send to Apollo successfully because in Apollo, the car could run well while I was operating the car manually in the simulation. But I don't know why car can't run automatically.
[image]

closed 2019-12-17 by kanbei