lgsvl/simulator #551: developer documentation?

AchimTuran, 2019-12-09:
Hi lgsvl development team,

honestly the simulator is an awesome project and it's great that it is based on Unity! Furthermore it seems to perfectly fit to the next steps of my project.

I do have some questions:
1. Is there a developer documentation?
2. Is there a possibility to add a custom bridge?
3. Where can I find the documentation about the JSON messages of the websocket API?

I'm looking forward to new features and discussions.

Thanks in advance for reading this issue and your support.

martins-mozeiko, 2019-12-09:
Most of our documentation you can find on our website: https://www.lgsvlsimulator.com/docs/

It is possible to add new bridge. Current implementation of bridges are cleanly separated - ROS from Cyber. So adding new one should be relatively straight forward. I suggest start looking at existing bridge implementation, for example, `CyberBridgeFactory.cs`, `CyberBridge.cs` and `CyberConversions.cs`. That should give you idea what needs to be implemented. Rest of simulator code - sensors, vehicles, etc.. are bridge agnostic. They work only with interfaces, thus they support any bridge.

By websocket API you mean rosbridge? Rosbridge protocol is documented here: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md

AchimTuran, 2019-12-09:
@martins-mozeiko 
Thank you for your quick response. 

Okay that's the documentation I already read. Yes these are the files, which I already identified and I wasn't sure if that is the correct place to get started. Seems to be that the implementation follows a clean architecture, which is great for extensions!

With websocket API I meant the already existing [Python API](https://www.lgsvlsimulator.com/docs/python-api/).

1. Where can I find the definitions of the existing JSON objects that are processed by the simulator?
2. Is it correct that the existing data objects are placed [here](https://github.com/lgsvl/simulator/tree/master/Assets/Scripts/Bridge/Data)? If yes these are all the data types a bridge needs to support right?

For my use case I will need a C/C++ binding, which is the reason why I asked the questions.

AchimTuran, 2019-12-09:
Okay then I will look deeper into [Assets/Scripts/Bridge/Cyber/CyberBridge.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberBridge.cs), [Assets/Scripts/Bridge/Cyber/CyberConversions.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberConversions.cs) and [Assets/Scripts/Bridge/Cyber/CyberBridgeFactory.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberBridgeFactory.cs).

AchimTuran, 2019-12-09:
Further question, is this the [place](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Database/DatabaseManager.cs#L223) to register the bridge factory?

martins-mozeiko, 2019-12-10:
1.

We do not have any public documentation on json communicated with Python API as it is meant to be private protocol, so we can change that for next versions (as long as we maintain stable Python API). You could check [Assets/Scripts/Api/Commands](https://github.com/lgsvl/simulator/tree/master/Assets/Scripts/Api/Commands) folder which has C# script for each different API command it receives. Main API processing loop is in [ApiManager.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Api/ApiManager.cs)

2. 

"Data" objects are message types that sensors can publish. Each bridge implementation must know how to convert specific Data class to its on message. This is what CyberConversion.cs or RosConversion.cs files are doing. Not all Data types are processed by all bridges. Bridge can choose to support only subset of Data types. That's why BridgeFactory class has list of supported Data types (`SupportedDataTypes` member). Currently it is not used, but later we will use this information to validate if set of sensors added to vehicle is able to talk to selected bridge.

3.

You don't need to register bridge factory. As long as you inherit from `IBridgeFactory` class, the simulator will be able to use it. Code automatically finds all classes that inherit IBridgeFactory by using C# reflection. It happens in following location: [Assets/Scripts/Utilities/BridgeTypes.cs#L24](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Utilities/BridgeTypes.cs#L24)

closed 2020-04-08 by EricBoiseLGSVL