lgsvl/simulator #552: updating from previous simulation version?

enisotomata, 2019-12-09:
Hello, is it possible to update the simulation version used in unity project, or must i recreate a different unity project for each simulation version?

martins-mozeiko, 2019-12-09:
I'm not sure what you mean by "created a project". Simulator is only one project, you cannot create new project and use Simulator.

If you mean that you created map or vehicle, then that should automatically work in newer simulator releases. There might be cases when C# script has changed, so some properties might need to be adjusted, but that's pretty much it.

enisotomata, 2019-12-09:
Yes, i have a map of a huge area and i wonder how can i use it with the updated unity simulator project.
i guess i need to rebuild the maps from unity?

martins-mozeiko, 2019-12-09:
If you are building maps to asset bundles to be used with binary version of simulator, then yes - you need to simply rebuild map in newest version of code.
If you are working with the map inside Unity Editor then there is nothing special to do, just open it and use it.

closed 2020-04-08 by EricBoiseLGSVL