lgsvl/simulator #553: build and run lgsvl and webUI in separate nodes

david-gwa, 2019-12-10:
hey, LG team, 
       
is there a way to build WebUI, so that can the lgsvl and webUI can run in two different machines? 

I am new with React/webUI,  there maybe some cross-domain issue to separate lgsvl server and browser client,  but I think there is some common solutions. 

Thanks
David

martins-mozeiko, 2019-12-10:
There is no need to build anything in special way. This kind of feature is already supported by default. All you need to do is edit `config.yml` file and specify which interface to bind to when listening for webui requests. Change `hostname` field either to ip address on your network, or just to `*` to listen on all interfaces. After this change you'll be able to open simulator web interface from different machine.

david-gwa, 2019-12-12:
thanks @martins-mozeiko , both the server side  config.yml need modify  `webHost` to a special IP address;  and  the `package .json` file from  WebUI/react project ,  need to modify its proxy to the same special IP address.

```
	"enact": {
		"theme": "moonstone",
		"proxy": "http://192.168.0.10:8080",
		"ri": {
			"baseSize": 16
		}
	},

```

david-gwa, 2019-12-12:
@martins-mozeiko 
I am afraid the modification as mentioned above, is not a whole solution,  as my remote node(192.168.0.11) has no access to Internet, but can access 192.168.0.10

* the first issue, I can' get account authority in the remote node 

* then, as all the assets (maps, vehicles) are stored as sqlite.db in the server host(192.168.0.10),  it is not accessible by the remote node 

[image]

what's your suggestion to deal with the asset in sqlite if want to remote access ?

Thanks again
David

martins-mozeiko, 2019-12-12:
I'm not sure why you need to modify packages.json onder WebUI folder. That only deals with building react stuff. It does not affect how web server will listen. Web server configuration is done in Unity .NET code. Currently it is controlled from settings in config.yml file. All you need to do is change hostname to "192.167.0.10" or "*". After that your *.11 machine should be able to access web interface on *.10. That's only change you need to do.

First time you access Simulator web interface from new browser it will return 401 - that is perfectly normal. Then it will redirect to account page for login. Not sure why there is no redirect happening. Can you show output of Console? There should not be any "undefined" errors popping up.

david-gwa, 2019-12-23:
thanks for replying, Martin.

the problem is the worker nodes has no access to Internet,  that may be the reason `redirect to account page` doesn't happen.  I tried to bypass  `this.RequiresAuthenticatin()` in  Unity Web/Modules, which did the trick.

closed 2020-04-08 by EricBoiseLGSVL