lgsvl/simulator #556: Add a new vehicel model as ego vehicel

Felixlxc, 2019-12-10:
Hi, I'm trying to add a new vheicle model in the simulator and add sensors to it. I followed the instruction of adding new ego vehicles. However, when I run the simulation, the sensor options are not allowed and shown on the UserUI panel. Can someone help me with this issues? Attached are screen shots for the new vehicle in the hierarchy panel and in the agent setup scripts.
[image]
[image]
[image]

EricBoiseLGSVL, 2019-12-10:
@Felixlxc 

I recommend that you update to the latest simulator version so you can implement new vehicles and sensors much easier.

For the UI in the old simulator, each sensor creates it's own UI element.  For example, RandarSensor.cs calls on Awake(), which creates the prefab and sets the listener

```
    private void AddUIElement()
    {
        if (Agent == null)
            Agent = transform.root.gameObject;
        var radarCheckbox = Agent.GetComponent<UserInterfaceTweakables>().AddCheckbox("ToggleRadar", "Enable RADAR:", IsEnabled);
        radarCheckbox.onValueChanged.AddListener(x => Enable(x));
    }
```

If you have a new UI with multiple options, you will need to create the prefab, add it to the Tweakables list and write the code in Tweakables to create this new UI element and listeners.

Again, I highly recommend updating to the latest simulator version.  We have made it much easier to add any type of sensor with parameters directly from JSON. Still, hope this helps.

Felixlxc, 2019-12-13:
@EricBoiseLGSVL Hi, thanks for your reply. But one thing that I'm confused about is that do I need to edit the script for each sensor if I add them to a new vehicle? I'm copying from an existing vehicle and refering to #198 for adding a new vehicle in it.

EricBoiseLGSVL, 2019-12-13:
@Felixlxc 

It looks like no sensor UI is working,  I think you have an issue setting up the vehicle.  It's odd that you have multiple sensors named "easy10_lite" and nothing showing up at run time.  Do you have any error logs you can share?

closed 2020-04-08 by EricBoiseLGSVL