lgsvl/simulator #557: Tool does not run on ubuntu 16.04

arufoOptimus, 2019-12-11:
When I run the executable in Ubuntu 16.04 the tool closes and nothing happens. This occurs if i double click the icon or if I run the code from the command line. Help would be most appreciated.

martins-mozeiko, 2019-12-11:
Have you installed libvulkan1?
https://www.lgsvlsimulator.com/docs/faq/#why-does-the-simulator-not-open-on-linux

arufoOptimus, 2019-12-11:
I will try that shortly and see what happens

> Have you installed libvulkan1?
> https://www.lgsvlsimulator.com/docs/faq/#why-does-the-simulator-not-open-on-linux

Fred159, 2019-12-19:
> Have you installed libvulkan1?
> https://www.lgsvlsimulator.com/docs/faq/#why-does-the-simulator-not-open-on-linux

It works for my situation. 
Thanks you ~

closed 2020-04-08 by EricBoiseLGSVL