lgsvl/simulator #559: the fied of view of the used sensor bigger than the max value of lgsvl simulator

weiyinxie, 2019-12-12:
hello as the titel says, the fov of my sensor is 90 degree and the center angle of my sensor is -45. according to to home page of lgsvl it's already over the max value. does any one have idea about it or if its possible to change the range of the possible value?

martins-mozeiko, 2019-12-12:
Current implementation of lidar won't handle very large filed of view for lidar, that will give too much distortion and precision issues.

You can try using large values and see how it looks like, but I expect it won't be good. There are no limits applied to sensor parameters you put in json - you can put arbitrary value there.

weiyinxie, 2019-12-13:
@martins-mozeiko hi! i have try to type a  value larger than limit but the result looks quiet different than the measurement in test field. So my boss ask me to modify the given sensor model so that it could also handle with the large value? Do u think is it possible for me to achieve it within one week? or whats the difficulty from your point of view?

weiyinxie, 2019-12-13:
@martins-mozeiko or if it's too difficult or impossible, can u tell me the reason why it's so difficult to achieve this goal so i can persuade my boss to change the idea asap

weiyinxie, 2019-12-13:
@martins-mozeiko or more precisely why is the limit of center angle set to the absolute value of 45 degree and the max fov set to 45 degree?

martins-mozeiko, 2019-12-13:
There is no real reason to have exactly limit of 45. You can easily increase it up to something like 60 and will still work. We just put values there that seemed reasonable - I don't think there are Velodyne models out there that has more than 45 degree field of view. But because lidar is implemented with camera, it cannot go much higher. Around 90 probably it will start giving too much imprecision. Fixing that would require changing way how lidar camera is rendering. For example, do rendering two times to capture only half of frustum in each rendering pass. Or rendering to cubemap which could give full 180 or even 360 view. It is all doable if you know how Unity rendering works with render textures and shaders.

I'm not sure what does not match to your measurements, but you can always tweak the values in code if they are wrong. There is a lot of stuff to adjust there.

Alternatively you could place two lidars with half of desired field of view and position them in a way which would give you large field of view together. It will publish two lidar messages not one, but that can be changed on ROS/Cyber side with small node that joins data together if needed.

weiyinxie, 2019-12-16:
@martins-mozeiko do u think is it possible to fix this proble by modifying the projection matrix

martins-mozeiko, 2019-12-16:
Changing just projection matrix won't help. That is actually core of the issue here. How projection of frustum works on GPU and how lidar behaves in real world does not match. For small angles the issue is not significant and it can be ignored, but for large angles the problem will be more noticeable and that's why it will require different approach.

weiyinxie, 2019-12-17:
thx, i have fixed the issue by changing the rotation matrix which requieres a very low Horizontal angle limit, otherwise it will be too noisy. But a low horiziontal angle limit consume a lot of computing power and leads to a high latency in my ros..........the hz of my lidar topic has been half cutted

closed 2020-10-06 by EricBoiseLGSVL