lgsvl/simulator #561: help: why i recv None

zyProgram, 2019-12-12:
I know that your simulation can only be connectted by one python client 。but i need to get senser's data and send command to car agent at the same time.  
    so I use two thread, ,one thread is reading date of sensors of car ,and another is sending comand to simulation . i found that this code 
  def get_sensors(self):
    j = self.remote.command("vehicle/sensors/get", {"uid": self.uid})
    return [Sensor.create(self.remote, sensor) for sensor in j]
have error:
   TypeError: 'NoneType' object is not iterable
 i dont know that why  i recv None when i get_sensors in one thread

martins-mozeiko, 2019-12-12:
Are you using Python API from multiple threads at same time? That is not supported. All API calls should come from same thread.

zyProgram, 2019-12-13:
yes , I used multiple thread.  
 your words means that lgsvl can be controlled by only one thread at one time, multiple thread may causes errors

martins-mozeiko, 2019-12-13:
Yes, that is correct.

I would suggest you to design system where all simulator calls are called on one separate thread if you want to avoid blocking main thread. And do communication from that separate thread to main thread by passing messages in queue or smth like that. Note that this means any callback from Simulator Python API like collision callback or waypoint reahed callback will be called on that separate thread.

closed 2020-04-08 by EricBoiseLGSVL