lgsvl/simulator #562: Jaguar2015XE Vehicle Dimensions

srirambadri, 2019-12-12:
Hi,

I am trying to get the raw sensor information into ROS and I've been successful in doing that.  

My question is, how do I find the dimensions of the vehicle? I tried to build the Jaguar2015XE Vehicle in Unity but I'm not able to find the dimensions. 

I basically want to fit many sensors eg. 4 lidar sensors instead of one placed at all four corners of the vehicle. I can add the sensors and with the transform option place them relative to each other using the configure option in the Web UI but where is the "base_frame" i.e 0,0,0 of the vehicle.

martins-mozeiko, 2019-12-12:
To get dimensions for Jaguar do the following:

1) Select `Body` game object (under Jaguar2015XEModel) in Hierarchy panel

2) In Inspector click `Add Component` button and choose `Box Collider`

3) This will add box collider that will encompass whole mesh of vehicle and also it will show sizes of box collider which you can use - (2.07, 1.27, 4.67).

4) Don't forget to remove this collider when you are done.

srirambadri, 2019-12-12:
@martins-mozeiko  thanks. I'm not able to figure out where the origin of the car. 
[image]

For example the camera sensor is placed at "x": 0, "y": 1.7, "z": -0.2, relative to what position?

martins-mozeiko, 2019-12-12:
It is in middle of car, at the bottom where ground is.

You can open car prefab and see its origin in Editor:
[image]

srirambadri, 2019-12-12:
perfect thanks

closed 2019-12-12 by srirambadri