lgsvl/simulator #563: win10 build from source  HDRenderPipeline error

muzi2045, 2019-12-13:
I an trying to build the lastest simulator code in unity 2019.1.10f1, this HDRenderPipeline error occur to me, hopefully for any advice.

[image]

martins-mozeiko, 2019-12-13:
These are not real errors, you can ignore them. Simply press "Clear" button and they will go away.

Sometimes Unity likes to complain about some kind of errors or warnings from their internal code. But everything still works just fine.

muzi2045, 2019-12-14:
after git lfs checkout, the whole repo are completed download, but still error occured, and it looks like can't running before resolve those compile errors(clear button help nothing )
[image]

muzi2045, 2019-12-16:
OK , By the way , I check the download files one by one , and find git lfs clone the repo can't clone complete files, After merge the git lfs file into git clone repo file,  it has no error occur to me.

closed 2019-12-16 by muzi2045