lgsvl/simulator #565: Autoware-Data creation for a custom map

mehmetdogru, 2019-12-13:
Hi,

I am trying to create a custom map in LGSVL and i wanna use it with autoware. Map folders in "_autoware-data_" have "_calibration/camera_lidar_3d/prius/_" folder which holds calibration for what purpose exactly? Do i need to create it in order to create the map compatible with autoware and the answer is yes then could you please tell me how?

Thanks a lot in advance for the help. You are doing a great job!

martins-mozeiko, 2019-12-13:
I think there is a bit of mess in this repository. Calibration data does not really belong to map. It should be specified for vehicle. It contains camera and lidar calibration data for vehicle. If you want to continue using vehicles we provide, simply copy&paste existing files and don't worry about changing them. I think we should move calibration folder outside of each map folder, so it is only in one copy in this repository.

mehmetdogru, 2019-12-14:
Ok, thank you.

closed 2019-12-14 by mehmetdogru