lgsvl/simulator #567: Apollo 5.0  migrated to Ubuntu 1804

ntutangyun, 2019-12-17:
Hi all, the Apollo 5.0 has migrated to Ubuntu 1804. and the cyber RT python wrapper also migrated to python3 instead with new library called cyber_py3, compared to existing cyber_py for python2. 

I'd like to ask if there is currently any effort to port the current lgsvl/apollo-5.0 fork to ubuntu 1804 version. 

If not, I'm willing to help out although I'm still quite new in this area. 
let me know how I could help and who I could work with to do the migration and build the new docker image with ubuntu 1804 base. 

Thank You.

martins-mozeiko, 2019-12-17:
We are currently not working on this, but having newer Apollo source code would be useful to have.
If you want to look into this, I suggest finding latest commit we have from original Apollo repo and checking history what was done afterwards to know what needs to be merged. Previously we have done git rebase to apply our commits on top of newer Apollo code base. It leads to some merge conflicts, but after manually fixing those everything worked fine.

ntutangyun, 2019-12-18:
@martins-mozeiko ok. I'll see what I can do during free time.

closed 2019-12-18 by ntutangyun