lgsvl/simulator #570: The simulator crashed when setting configuration of camera  & Perception and traffic light can't be enable

kanbei, 2019-12-17:
Hello, I have 3 questions.
The first one is if I set JSON Configuration of camera, according to the example(https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/), the simulator was crashed suddenly after I created the simulator and clicked 'play' button.
When I deleted the configuration of camera, the simulator worked well. How could I set the configuration of camera?

The second one is the perception and traffic light module can't be enabled in Apollo. I tried to click them but soon after it became unable.

The third questioin is when I added point and send routing request, the car was still in the initial position, didn't run automatically.
I think the position information in the simulator can be send to Apollo successfully because in Apollo, the car could run well while I was operating the car manually in the simulator. But I don't know why car can't run automatically.
[image]

martins-mozeiko, 2019-12-20:
When simuator crashes, please check the logfiles. At bottom there may be more useful information about what happened.

If crash happens only when you add cameras, it could be that your GPU runs out of memory. Monitor your GPU usage when launching simulator, to check if this is an issue. If next upcoming release we ave signficantly optimized required memory on Borregas and other maps.

Percetion and Traffic light modules do not work, beause they are not receiving camera image. This is expected. Once camera publishing will work, these modules will stay on.

If car does not start automatically, try "helping" it a bit - press up-arrow key for car to start drive. After that it should start driving automatically. Sometimes we see this behavior from Apollo.

ntutangyun, 2020-01-09:
@kanbei Another way I found useful to debug is to use the Unity Editor to launch the simulator. When it crashes, the console tab is expected to log the errors.

martins-mozeiko, 2020-01-09:
There's no need to run Unity Editor to see log messages. Simulator binary produces same log messages in logfile: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

ntutangyun, 2020-01-10:
@martins-mozeiko 👌

closed 2020-04-08 by EricBoiseLGSVL