lgsvl/simulator #571: Multiple Egos in Simulator

arslansarp, 2019-12-17:
Hello, I just want to control multiple ego vehicle in simulator. How can I add other vehicles and control them. 

Could it be done with just creating in it run time. Attach vehicle controller in it? Please give me guidance if it supports

martins-mozeiko, 2019-12-17:
If you launch Simulation from Web interface, then simple add multiple vehicles in simulation page: https://www.lgsvlsimulator.com/docs/simulations-tab/
If you are running Python API you can instantiate multiple agents with EGO type.

At runtime you'll be able to switch between EGO vehicles with keyboard 1...9 numbers or at bottom-right corner of screen.

arslansarp, 2019-12-17:
First of all thank you for answer. But, currently I am using old version of lgsvl and I could control vehicle at runtime with some topic that outside of the simulator. I wonder , can  I modify the controllers of the ego vehicles that have been added on web ui at run time. Currently I have control on one of them.( using old version)

arslansarp, 2019-12-17:
I just want to give steer, gas, brake and so on at runtime. Not with manual control for all other ego vehicles

martins-mozeiko, 2019-12-17:
You need to connect each vehicle to different rosbridge. Then each vehicle can connect to same topics. As long as they will run with different rosmaster nodes the AD environments will be isolated and you will be able to control each EGO vehicle independently.

arslansarp, 2019-12-18:
Hello again martin, I added several vehicles , it works perfect. But I wonder how can I change the position of the start. I meant when I add three vehicle in simulator, they just stacked up. All of them spawn in the same position. How can I change their default spawn position one by one .

OutOfCage, 2019-12-20:
If you don't use Python API, I have two temporary methods.
1. Add more spawns on map scene.  You can copy spawn and change transform.
[image]
2. You can change the code of  function `CreateAgentsFromConfigs` in  `AgentManager.cs`. 

Hope for better solutions in the future. If you can good ideas, we can discuss.

closed 2020-04-08 by EricBoiseLGSVL