lgsvl/simulator #573: How to change frequency of main camera

clepz, 2019-12-19:
Hello,
I need to change frequency of main camera. How can I do that? I tried to change fps in json but nothing happened. It always publishes at 10hz

OutOfCage, 2019-12-20:
Maybe the sensor script uses `Update` to publish message and upper frequency limit is 10HZ.  You can try to change `Update` to `FixedUpdate`.  I suggest you can have a look at [Unity Time](https://docs.unity3d.com/ScriptReference/Time.html) before you do some change.

martins-mozeiko, 2019-12-20:
Because camera rendering is using GPU, its publishing frequency will be affected of graphics performance of Simulator. Make sure they frames-per-second metric for our Simulator on your machine is reasonably high - 40 or 50 at least. Only then you'll be able to get higher camera publishing frequency.

clepz, 2019-12-20:
@martins-mozeiko I have 1080ti. It's at 10 hz anyway.

martins-mozeiko, 2019-12-28:
What framerate do you get for main view of simulator?

You should try running with as simple as possible configuration to eliminate other potential performance bottlenecks - try running SingleLaneRoad map with vehicle that has only Camera sensor + vehicle control sensor. Remove all other Camera or Lidar sensors. Then check the publishing rate.

closed 2020-04-08 by EricBoiseLGSVL