lgsvl/simulator #576: having build problems?

enisotomata, 2019-12-20:
hello, i have some issues building assetBundles from unity. after having a successful build , if scene fails to build after adding some extra models how can i restore my previous successful setup. i tried to copy project after a working build but it does not build again. Is it possible to ruin whole project by copyin and pasting back?

SeymurSm, 2019-12-21:
Hi. It seems you get empty assetBundle after build (Something around 500kb in file size).  One of the reason should be to cause of fail is that you should add your asset to specific folder. For example if you are building environment assetBundle you should place all your environment assets inside of **Environment/SceneName/models**   folder.  I thing it is not possible to restore previous successful build. You should keep your builds with version number or something like this.

enisotomata, 2019-12-21:
i have also prefabs folder with models folder and i prefab every model in scene even it will be used again or not and i carry prefabs to prefabs folder. if a successful build is 2.4 gb failing build wlth same scene is mostly 500 mb not kb.

martins-mozeiko, 2019-12-28:
How exactly scene fails to build? What are the error messages?

EricBoiseLGSVL, 2020-04-08:
@enisotomata Your scene is too large.  You need to reduce mesh, textures and size.  Do not copy and paste files in/out of unity project because it can corrupt meta files.

closed 2020-04-08 by EricBoiseLGSVL