lgsvl/simulator #577: Radar not detecting vehicles

arslansarp, 2019-12-23:
Hello,

Radar is detecting objects in lgsvl map (shalun,singlelane, exc.. ) . However, in our custom map, when I add some vehicles on map, it isn't detecting them. I assume it might be the collider issue but I have already added collider mesh. The car can't pass through the vehicles. 

[image]
[image]

martins-mozeiko, 2019-12-23:
How are you adding vehicles to map? If this is your own custom code or you are simply drag&drop'ing them on the scene, then this is expected. When NPC's are added to scene, there is an extra setup required for ground truth sensors to work. Relevant code is here: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Managers/NPCManager.cs#L117

arslansarp, 2019-12-23:
Thanks for your answer martin. As you said, I just drag&drop them. Is there any shortcut for this setup?
I mean can I handle it in Unity Editor, if so , I will be pleased if you help me with a clear guidance.

arslansarp, 2019-12-24:
Hey Martin , I just solved the issue. Thank you again..

[image]

pushkar4501, 2019-12-25:
@arslansarp Hi, can you help me through the process of adding traffic in the customized map?
thanks

closed 2019-12-28 by martins-mozeiko