lgsvl/simulator #578: lgsvl suddenly closed when run apollo 5.0 in dreamviewer.

Fred159, 2019-12-24:
I tried to co-simulate the lgsvl with apollo 5.0. However, when dreamviewer start, lgsvl simulator close suddenly. I don't know the reason. How to solve this ? 
Is this because of hardware? 
My hardware is 8th i7, 12GB memory, gtx1060.
The problem is recorded in below link.
https://www.youtube.com/watch?v=4zIcKuwbLmY&feature=youtu.be

cyberphysicalrobotics, 2019-12-26:
Try nvtop to check gpu memory usage. When simulator crash happens, please check the gpu memory usage. Let me know the result.

https://github.com/Syllo/nvtop

martins-mozeiko, 2019-12-28:
@Fred159 If this happens on BorregasAve map, then try updating to latest 2019.12-rc1 version. It has new map that significantly reduces required memory.

Fred159, 2020-01-22:
> @Fred159 If this happens on BorregasAve map, then try updating to latest 2019.12-rc1 version. It has new map that significantly reduces required memory.

Thank you . I am tring to use better computer. I will try it and update .

closed 2020-01-22 by Fred159