lgsvl/simulator #581: Radar not detect npc in Waypoints Control.

OutOfCage, 2019-12-25:
Hi, LG Team.
I find an issue that radar can not detect npc when npc is under Waypoints control. I think [this](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Controllers/NPCController.cs#L266) may cause the issue.
I use Python API to add npc on scene. In Unity Editor, I choose `isTrigger` of npc meshcollider and `isKinematic` of npc rigidbody. Then this issue arises.

OutOfCage, 2019-12-25:
Also, when I use Python API to add npc on scene without waypoints, the radar can detect the npc. But after I add waypoints to follow, the radar does not detect the npc.

EricBoiseLGSVL, 2020-10-06:
For triggers to work, isKinematic must be false, this will not work

closed 2020-10-06 by EricBoiseLGSVL