lgsvl/simulator #589: How to directly control EGO vehicle through API

xiao-hua-sheng, 2020-01-08:
Hi
I want to use the API to control the EGO vehicle without using ros, Apollo and other systems.Currently I can load the map and EGO vehicle through the API, but I can't bridge the EGO.The official routine is PythonAPI-master\quickstart\22-connecting-bridge.py
[image]
[image]
[image]
[image]

martins-mozeiko, 2020-01-08:
"Bridge" is meant for connecting our simulator to ROS or CyberRT nodes. If you don't want to use Apollo or any other ROS/Cyber environment, then there is no need to connect EGO vehicle to bridge.

xiao-hua-sheng, 2020-01-08:
> “桥接器”用于将我们的模拟器连接到ROS或Cyber​​RT节点。如果您不想使用Apollo或任何其他ROS / Cyber​​环境，则无需将EGO车辆连接到桥梁。

Thank you for your reply.
How did the control signal predicted by the model pass to the EGO vehicle?
Can't I control the EGO vehicle in real time directly through the API?You have to have support from other platforms

martins-mozeiko, 2020-01-08:
If you want to control EGO vehicle in real-time you should use either ROS or Cyber messages. This will require running bridge. Python API is not designed for real-time control. It should be used only to set up the scene - setting initial position, adding other NPC vehicles or pedestrians.

To control EGO vehicle from ROS or Cyber, you should publish either Autoware, Apollo 3.0 or Apollo 5.0 control message with necessary information (throttle, braking, steering). If you want to send custom control message, that will require modifying Simulator source code to listen for custom command.

closed 2020-04-08 by EricBoiseLGSVL