lgsvl/simulator #590: Apollo not able to change lane

ntutangyun, 2020-01-09:
Hi team, 

I'm using LGSVL's Apollo 5.0 fork and shalun map.

I found that Apollo is not able to change lanes. It could only navigate through the connected lanes but cannot change lanes, even though the routing_map.bin has information that lane changing is allowed: e.g. leftOutList, rightOutList etc.

I'd like to ask if anyone knows if it's Apollo's own issue or something else ?

Thanks.

luqiang21, 2020-01-09:
@ntutangyun If Apollo is not able to change lanes, you'd better check the lane information of the interested lane. It might be the case that neighbor lanes information is not exported correctly for that lane, thus Apollo doesn't know that lane has a left/right lane to change to.

To check the map file, you need to convert `base_map.bin` to `base_map.txt` to view it. Inside apollo docker, you can convert this way:
`protoc --decode=apollo.hdmap.Map modules/map/proto/map.proto --proto_path=. < /apollo/modules/map/data/Your_map_folder/base_map.bin > base_map.txt`

Then find out the laneId of your lane using apollo map tools, look for it inside the `base_map.txt` file.

ntutangyun, 2020-01-10:
@luqiang21 Thanks for your prompt reply.

Here's an example in Shalun Map: lane 28 and lane 29
(both base_map.bin and routing_map.bin downloaded from lgsvl content website)

[image]

[image]

From **base_map.bin**, the only information relevant for lane change between 28 and 29 is:

lane 28 -> rightBoundary -> boundaryType = 2  (which is DOTTED WHITE).  
lane 29 -> leftBoundary -> boundaryType = 2 (also DOTTED WHITE).

According to my understanding and intuition, I think DOTTED WHITE here means can change lane without problem.

from **routing_map.bin**, the only information relevant for lane change between 28 and 29 is:

lane 28: right_out (all along the lane)
lane 29: left_out (all along the lane)

I think intuitively they means that you can lane change out from left or right if right_out or left_out is defined respectively.

Thus my assumption here is that according to the base_map.bin and routing_map.bin, Apollo should know that it can perform lane changing between lane 28 and lane 29. 

However during actual simulation, (Apollo 5.0 LGSVL forked version), it turns out that 
if Apollo vehicle is at start of Lane 29 and asked to stop at lane 28, it will make some big circles around the map instead of simply lane changing to it..

[image]

is there any other information that's missing? Thank you :)

luqiang21, 2020-01-10:
@ntutangyun Thank you for your reply. I believe some information like `left_neighbor_forward_lane_id` is missing from the map file. It could be a bug in our simulator map exporter. You can check [our previous map file](https://raw.githubusercontent.com/lgsvl/apollo-3.0/7840d909f64dff6c38eeed90dd382fea607202e4/modules/map/data/shalun/base_map.txt), we had that information. We will look into this and fix it.

ntutangyun, 2020-02-10:
@luqiang21 Hi, I like to follow up with the case, whether the maps are fixed. Thank You.

luqiang21, 2020-02-10:
@ntutangyun We fixed this in the most recent [release](https://github.com/lgsvl/simulator/releases). Please try and let me know if it doesn't work.

ntutangyun, 2020-02-13:
@luqiang21 tested borregas Ave map. seems that the vehicle can perform lane change now. 
Thank You.

closed 2020-02-13 by ntutangyun

gaohongfein, 2020-03-27:
@luqiang21 @ntutangyun  Hi team, have you fix the bug in annoation map? When we annotate the map in Unity, can the vehicle change the lane? have you tested?

ntutangyun, 2020-03-27:
@gaohongfein hi, in the later updated maps, the lane relationships are restored. and Now Apollo is able to change lanes where possible.

JasonBourne1998, 2022-06-14:
@ntutangyun Hi, I try the SanFrancisco map, but it still could not change lane, could you give some suggestions about it?

ntutangyun, 2022-06-15:
Hi @JasonBourne1998 YiFan, I'm from NTU. you may add me on wechat: XX_101n374i1_XX

JasonBourne1998, 2022-06-15:
> Hi @JasonBourne1998 YiFan, I'm from NTU. you may add me on wechat: XX_101n374i1_XX

awesome! I have added you, waiting for your agreement.:)