lgsvl/simulator #591: Asset Bundle Build Error

OutOfCage, 2020-01-10:
Hi, LG Team.
Last month, we change the version from 2019.07 release to 2019.11.  We check many functions work but sometimes build map or vehicle asset bundles error.  This is the error log.
```
FileNotFoundException: Could not find file "/home/moonx/Desktop/new/AssetBundles/11c59c48-2d97-4056-ad71-ef7ac10da9f5_environment_textures"
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <7d97106330684add86d080ecf65bfe69>:0)
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
(wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare)
System.IO.File.Open (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
ICSharpCode.SharpZipLib.Zip.StaticDiskDataSource.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.AddEntry (ICSharpCode.SharpZipLib.Zip.ZipFile workFile, ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate update) (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.RunUpdates () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.CommitUpdate () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
Simulator.Editor.Build.RunAssetBundleBuild (System.String folder, System.Collections.Generic.List`1[T] environments, System.Collections.Generic.List`1[T] vehicles) (at Assets/Scripts/Editor/Build.cs:400)
Simulator.Editor.Build.OnGUI () (at Assets/Scripts/Editor/Build.cs:219)
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) (at <7d97106330684add86d080ecf65bfe69>:0)
UnityEditor.HostView.Invoke (System.String methodName, System.Object obj) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:340)
UnityEditor.HostView.Invoke (System.String methodName) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:334)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at /home/builduser/buildslave/unity/build/Editor/Mono/HostView.cs:310)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect, System.Boolean floatingWindow, System.Boolean isBottomTab) (at /home/builduser/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:373)
UnityEditor.DockArea.OldOnGUI () (at /home/builduser/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:340)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:295)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:481)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:464)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at /home/builduser/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:444)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/Events/MouseCaptureDispatchingStrategy.cs:93)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:280)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at /home/builduser/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:156)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at /home/builduser/buildslave/unity/build/Modules/UIElements/Panel.cs:189)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at /home/builduser/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:255)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at /home/builduser/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:78)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at /home/builduser/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:179)
```
In my code, `Assets/Scripts/Editor/Build.cs:219` is [L264](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Build.cs#L264) in current code.
```
                    RunAssetBundleBuild(assetBundlesLocation, environments.ToList(), vehicles.ToList());
```
`Assets/Scripts/Editor/Build.cs:400` is [L447](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Editor/Build.cs#L447).
```
                            archive.CommitUpdate();
```
What cause this issue? How can I avoid this issue?

[image]
Thanks.

SeymurSm, 2020-01-10:
Hi, 
I thing you forget adding new assets files to required folder. For example, if you are building new Environment Asset bundle you should place all asset files on specific folder in ***/Assets/External/Environments/EnvironmentName/Models/EnvironmentName/*******.
For example lets say we have terrain on our new environment and we have applied texture assets on terrain. In such cases, we mush place all these texture assets, trees or whatever  in mentioned folder. 

Another reason may cause build error should be the error on scripts such as importing not project libraries.  Hope this helps.

Seymur

mark-gerow-lge, 2020-01-10:
SeymurSm is correct, all map specific assets should be in the subfolder to be built into bundles correctly. Otherwise it may detect no textures to build which will fail to add when we construct our vehicle/map bundles.

That said, you mention this only happens sometimes. Are you able to correctly build a vehicle then fail the next time? Or are some of your vehicles building and some not?

OutOfCage, 2020-01-14:
@SeymurSm @mark-gerow-lge Thanks for help.
Some of my maps or vehicle build fail and their asset files are not all placed  on specific folder. I will check this issue.

closed 2020-01-14 by OutOfCage