lgsvl/simulator #595: Start LGSVL as API only

MaxandreOgeret, 2020-01-14:
Hello everyone.

I would like to make my developments faster and for this I would like to setup my LGSVL simulation faster. I think it's very long to double click one the executable, open the browser, ... and so on.

For now I use the API mode to load the scene, create the ego vehicle, setup the bridge, ...

But I still have to press "Play!" is the configuration prompt, start the browser ans select the API mode.

Is there a way, like a config file or a command line parameter to start LGSVL already in API mode ?

Thanks a lot !

martins-mozeiko, 2020-01-14:
Currently there isn't any way to run API mode automatically from command line. What you could do is to send HTTP request automatically to simulator to run specific Simulation. Basically check what kind of HTTP API does browser send to simulator process when you click "Play" button in browser, and repeat it either from script or command-line tool (such as curl).

MaxandreOgeret, 2020-01-15:
Thanks for your answer ! I guess it will not be that easy as I would need to :

 - Get rid of the configuration window (This [one](https://answers.unity.com/storage/temp/53027-unity-configuration-screen.png)). How can I do that ? Do I need to recompile it ?
- Send the requests to the server. (Maybe using selenium or something ?)

In my opinion this should definitely be implemented as the whole purpose of an API mode is to setup everything from a script.

Thanks for your help !

EricBoiseLGSVL, 2020-04-08:
@MaxandreOgeret Configuration window is removed on next release, in the next few days.

closed 2020-04-08 by EricBoiseLGSVL