lgsvl/simulator #598: Failed to build Sensor plugin on linux

HansRobo, 2020-01-15:
build fails with below error.
```
Failed to find Windows asset bundle of ComfortSensor. Please correct other errors and try again.
```

it seems that environment check of windows is executed in linux build.

martins-mozeiko, 2020-01-15:
Do you see any other error messages above this?
Usually this error means that you did not choose Windows support for Unity during installation, so build cannot produce Windows targeted files.

HansRobo, 2020-01-15:
Certainly I did not select Windows support when installing. 
However, it is strange that the environment check of windows is performed when the build target is linux.

HansRobo, 2020-01-15:
Please wait a little longer for error confirmation

HansRobo, 2020-01-15:
Other errors are below
```
Building an AssetBundle for target 'WindowsStandaloneSupport' is not allowed because the required module is not installed.
```
[image]

martins-mozeiko, 2020-01-15:
The platform selection in Build dialog is only used when building simulator executable binary.

Assets - maps, vehicles and sensor plugins - are always built for both platforms: Windows and Linux. This is because when these plugins are shared with other users, they might run it on different platform than where they are built.

HansRobo, 2020-01-15:
Oh, I see.
But, I want documentation that describes it.
Then, number of people who falls into the same situation as me will decrease.

shalinmehtalgsvl, 2020-01-15:
Hi @HansRobo ,

Our build instructions do say to install Windows and Linux support. We'll update the docs to clarify the platform selection in the dialog.

https://www.lgsvlsimulator.com/docs/build-instructions/

HansRobo, 2020-01-15:
Thank you for your replay
> Our build instructions do say to install Windows and Linux support.  

Oh, I overlooked it !
I'll install windows support later.
> We'll update the docs to clarify the platform selection in the dialog.  

That's great.

closed 2020-04-08 by EricBoiseLGSVL