lgsvl/simulator #599: rviz cannot show vehicle pose in the instruction lgsvl with autoware.ai 1.12

DarrenWong, 2020-01-15:
Hey.

I am trying to setup the simulator with autoware.ai following below instruction, https://www.lgsvlsimulator.com/docs/autoware-instructions/

operation system: ubuntu16.04 with GTX1070
Autoware version: 1.12.0
simulator version: lgsvlsimulator-linux64-2019.11

I think i stuck in the below procedure and try many times and cannot get localization. Thanks a lot.
```
Then click "Rviz" to launch Rviz - the vector map and location of the vehicle in the map should show.

The vehicle may be mis-localized as the initial pose is important for NDT matching. To fix this, click "2D Pose Estimate" in Rviz, then click an approximate position for the vehicle on the map and drag in the direction it is facing before releasing the mouse button. This should allow NDT matching to find the vehicle pose (it may take a few tries). Note that the point cloud will not show up in rviz until ndt matching starts publishing a pose.
```
screenshot of current steps
0. autoware.ai version
[image]

1. lgsvl is connected and publish msg
[image]
2. autoware runtime manager
[image]

3. ros rviz only see map, i use defalut rviz config
[image]

4. wait some times and try many times 2D pose estimate, the status still not change
[image]

martins-mozeiko, 2020-01-16:
Are you setting initial position in middle of two yellow lines? Because that's not correct car position. In Simulator you can see that car is to the right of both yellow lines. Try setting initial position in second line from left side, not in middle of yellow lines.

DarrenWong, 2020-01-17:
thanks for your replay @martins-mozeiko , it seems my mistake for the wrong init pose

With your suggestion with a few try(in second lane, in lane boundary, or first lane, or some places up and down) still cannot get the init pose, would you mind giving a example screenshot for the correct  pose estimate?  And if the map have a marker(such as direction signal) for the init pose it would be very helpful for the quickstart example

Some init pose i tried,  and i noticed the point_raw in rviz status error but i can receive the point_raw msg by __rostopic echo /points_raw__
[image]
[image]
[image]
.

DarrenWong, 2020-01-17:
BTW when i start simulation it says my map out-of-date, so i update the map assets here, did it have any effects? actually i try latest 2019.12-rc1(with below content updated) also similar problems...
https://content.lgsvlsimulator.com/maps/borregasave/

https://assets.lgsvlsimulator.com/513316ab0ed5896f91cf7e6ff4e138b068993626/environment_BorregasAve

DarrenWong, 2020-01-28:
closed as no further response...

closed 2020-01-28 by DarrenWong

yugushihuang, 2020-02-17:
@DarrenWong 

I got exactly same problem as yours, did you get it resolved?

DarrenWong, 2020-02-18:
@yugushihuang  no..  tries many many times...

AIasd, 2020-02-18:
I encountered the same problem (system: ubuntu18.04 GPU: RTX2080TI). Does anyone solve it?

JockTokk, 2020-04-30:
@AIasd try to run the simulation in simulator and try to init position