lgsvl/simulator #600: Doubts about setting NPCs on the map

daohu527, 2020-01-16:
https://github.com/lgsvl/simulator/blob/48650e7eadff7a4a705fc03866bdb80a827c49b8/Assets/Scripts/Managers/NPCManager.cs#L221

We know that the spawn point is always the starting point of the lane. If the starting point is not on the screen, the car will not appear. That is, you cannot find NPC car until the ego car is near the end of the opposite lane. An example is the singleLane scene, there is no car in the opposite lane for a long time.
I think this limitation can be eliminated because "WithSpawnArea" is already defined.
Or we need change the spawn point. Maybe we should try some other point but not the lane start point.

shalinmehtalgsvl, 2020-01-16:
Hi @daohu527 ,

The area that NPCs can spawn in is limited for performance reasons. If you open the Map in Editor and look at the MapOrigin. In Inspector you'll find an `NPC Spawn Bounds Size` which is the radius around the EGO that NPCs will spawn in. The slider is limited to 300, but you can increase the limits in code if needed. `WithinSpawnArea()` uses the radius set in `MapOrigin`.

Also note that NPCs will not spawn in view of certain sensors including cameras. See [here](https://www.lgsvlsimulator.com/docs/sensor-plugins/) for a brief description of the `CheckVisible` function of sensors.

We're investigating a better solution for this.

daohu527, 2020-01-17:
@shalinmehtalgsvl
Thanks for your explanation and hope for a better solution.

EricBoiseLGSVL, 2020-04-08:
@daohu527 we have set NPCs and Pedestrians to spawn anywhere on first frame and at the start of annotations with no overlapping connections.  Hopefully this is a better solution.

closed 2020-04-08 by EricBoiseLGSVL