lgsvl/simulator #603: Simulator does not start on Nvidia M1000M, Ubuntu 18.04.

takanotume24, 2020-01-17:
Hello.
I tried launching this simulator on HP Zbook Studio G3, but did not.
When I tried it, it started up to the setting screen such as resolution, but if you click OK in the dialog, the screen will darken and after a certain time the Ubuntu login screen will be displayed. When you log in, the desktop will be displayed The simulator does not start.
The  ``Player.log``and ``vulkaninfo`` are described below.
```
Desktop is 3840 x 2160 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=65
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[... 845 lines cut ...]
VkPhysicalDeviceUniformBufferStandardLayoutFeaturesKHR:
-------------------------------------------------------
	uniformBufferStandardLayout = true

VkPhysicalDeviceVariablePointersFeatures:
-----------------------------------------
	variablePointersStorageBuffer = true
	variablePointers              = true

VkPhysicalDeviceVertexAttributeDivisorFeaturesEXT:
--------------------------------------------------
	vertexAttributeInstanceRateDivisor     = true
	vertexAttributeInstanceRateZeroDivisor = true

VkPhysicalDeviceVulkanMemoryModelFeaturesKHR:
---------------------------------------------
	vulkanMemoryModel                             = true
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = true

VkPhysicalDeviceYcbcrImageArraysFeaturesEXT:
--------------------------------------------
	ycbcrImageArrays = true

```

martins-mozeiko, 2020-01-17:
It seems that Xorg server is crashing when Simulator tries to use Vulkan.
Can you please check Xorg server log file after it crashes, maybe it it will have more useful error message. Check also kernel log messages (dmesg) which could have also something useful.

esteve, 2020-01-20:
@bshin-lge pointed me to this ticket via https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/issues/222#note_273342327

Here are the contents of my `Xorg.0.log` file:

```
[104984.352] (--) Log file renamed from "/var/log/Xorg.pid-25331.log" to "/var/log/Xorg.0.log"
[104984.352] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[104984.352] Build Operating System: Linux 4.9.0-8-amd64 x86_64 Debian
[104984.352] Current Operating System: Linux haggis 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64
[104984.352] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.19.0-6-amd64 root=UUID=12339cac-ded9-417c-bbf8-8d5636f566ea ro quiet
[104984.352] Build Date: 05 March 2019  08:11:12PM
[104984.352] xorg-server 2:1.20.4-1 (https://www.debian.org/support) 
[104984.352] Current version of pixman: 0.36.0
[104984.352] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[104984.352] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[104984.352] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jan 16 16:41:07 2020
[104984.352] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[104984.352] (==) No Layout section.  Using the first Screen section.
[104984.352] (==) No screen section available. Using defaults.
[104984.352] (**) |-->Screen "Default Screen Section" (0)
[104984.352] (**) |   |-->Monitor "<default monitor>"
[104984.353] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[104984.353] (==) Automatically adding devices
[104984.353] (==) Automatically enabling devices
[104984.353] (==) Automatically adding GPU devices
[104984.353] (==) Max clients allowed: 256, resource mask: 0x1fffff
[104984.353] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[104984.353] 	Entry deleted from font path.
[104984.353] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/100dpi/:unscaled,
	/usr/share/fonts/X11/75dpi/:unscaled,
	/usr/share/fonts/X11/Type1,
	/usr/share/fonts/X11/100dpi,
	/usr/share/fonts/X11/75dpi,
	built-ins
[104984.353] (==) ModulePath set to "/usr/lib/xorg/modules"
[104984.353] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[104984.353] (II) Loader magic: 0x563488b80e20
[104984.353] (II) Module ABI versions:
[104984.353] 	X.Org ANSI C Emulation: 0.4
[104984.353] 	X.Org Video Driver: 24.0
[104984.353] 	X.Org XInput driver : 24.1
[104984.353] 	X.Org Server Extension : 10.0
[104984.353] (++) using VT number 7

[104984.353] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[104984.354] (II) xfree86: Adding drm device (/dev/dri/card0)
[104984.355] (--) PCI:*(1@0:0:0) 10de:13c2:1462:3160 rev 161, Mem @ 0xee000000/16777216, 0xd0000000/268435456, 0xe0000000/33554432, I/O @ 0x0000e000/128, BIOS @ 0x????????/131072
[104984.355] (II) LoadModule: "glx"
[104984.355] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[104984.356] (II) Module glx: vendor="X.Org Foundation"
[104984.356] 	compiled for 1.20.4, module version = 1.0.0
[104984.356] 	ABI class: X.Org Server Extension, version 10.0
[104984.356] (II) Applying OutputClass "nvidia" to /dev/dri/card0
[104984.356] 	loading driver: nvidia
[104984.576] (==) Matched nvidia as autoconfigured driver 0
[104984.576] (==) Matched nouveau as autoconfigured driver 1
[104984.576] (==) Matched nv as autoconfigured driver 2
[104984.576] (==) Matched modesetting as autoconfigured driver 3
[104984.576] (==) Matched fbdev as autoconfigured driver 4
[104984.576] (==) Matched vesa as autoconfigured driver 5
[104984.576] (==) Assigned the driver to the xf86ConfigLayout
[... 498 lines cut ...]
[427434.791] (EE) unw_get_proc_name failed: no unwind info found [-10]
[427434.791] (EE) 4: /usr/lib/xorg/modules/drivers/nvidia_drv.so (?+0x0) [0x7f2174923290]
[427434.791] (EE) 5: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0xa9783) [0x7f2174a9c8d3]
[427434.792] (EE) 6: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0xaa91b) [0x7f2174a9e51b]
[427434.792] (EE) 7: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0x837f3) [0x7f2174a50903]
[427434.793] (EE) 8: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0x86eae) [0x7f2174a5734e]
[427434.793] (EE) 9: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0x8c51a) [0x7f2174a6228a]
[427434.794] (EE) 10: /usr/lib/xorg/modules/drivers/nvidia_drv.so (nvidiaAddDrawableHandler+0x6cd3e) [0x7f2174a234ce]
[427434.795] (EE) 11: /lib/x86_64-linux-gnu/libXext.so.6 (nvidiaAddDrawableHandler+0x4f9896) [0x7f217533cbfc]
[427434.795] (EE) 
[427434.795] (EE) Floating point exception at address 0x7f2174922f7a
[427434.795] (EE) 
Fatal server error:
[427434.795] (EE) Caught signal 8 (Floating point exception). Server aborting
[427434.795] (EE) 
[427434.795] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[427434.795] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[427434.795] (EE) 
[427435.083] (EE) Server terminated with error (1). Closing log file.
```

I did some research and found this Debian ticket https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939839 I have nvidia-418 installed, I'll try again with nvidia-435

takanotume24, 2020-01-20:
Thank you for the investigation.
I send Xorg Server logs and dmesg.
``/var/log/Xorg.0.log.old``
```
[     7.512] (--) Log file renamed from "/var/log/Xorg.pid-1092.log" to "/var/log/Xorg.0.log"
[     7.513] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[     7.513] Build Operating System: Linux 4.4.0-148-generic x86_64 Ubuntu
[     7.513] Current Operating System: Linux takuya-HP-ZBook-Studio-G3 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64
[     7.513] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.0.0-37-generic root=UUID=52f68d66-3ea2-4656-a06f-b01beccac52b ro quiet splash vt.handoff=1
[     7.513] Build Date: 02 May 2019  08:06:54AM
[     7.513] xorg-server-hwe-18.04 2:1.20.4-1ubuntu3~18.04.1 (For technical support please see http://www.ubuntu.com/support) 
[     7.513] Current version of pixman: 0.34.0
[     7.513] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[     7.513] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.513] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Jan  4 21:54:49 2020
[     7.514] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.514] (==) No Layout section.  Using the first Screen section.
[     7.514] (==) No screen section available. Using defaults.
[     7.514] (**) |-->Screen "Default Screen Section" (0)
[     7.514] (**) |   |-->Monitor "<default monitor>"
[     7.515] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[     7.515] (==) Automatically adding devices
[     7.515] (==) Automatically enabling devices
[     7.515] (==) Automatically adding GPU devices
[     7.515] (==) Automatically binding GPU devices
[     7.515] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     7.515] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     7.515] 	Entry deleted from font path.
[     7.515] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     7.515] 	Entry deleted from font path.
[     7.515] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     7.515] 	Entry deleted from font path.
[     7.515] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     7.515] 	Entry deleted from font path.
[     7.515] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     7.515] 	Entry deleted from font path.
[     7.515] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[     7.515] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.515] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.515] (II) Loader magic: 0x557ee5562020
[     7.515] (II) Module ABI versions:
[     7.515] 	X.Org ANSI C Emulation: 0.4
[     7.515] 	X.Org Video Driver: 24.0
[     7.515] 	X.Org XInput driver : 24.1
[     7.515] 	X.Org Server Extension : 10.0
[     7.516] (++) using VT number 1

[     7.517] (II) systemd-logind: took control of session /org/freedesktop/login1/session/c2
[     7.519] (--) PCI:*(1@0:0:0) 10de:13b1:103c:8108 rev 162, Mem @ 0xe3000000/16777216, 0xa0000000/268435456, 0xb0000000/33554432, I/O @ 0x00003000/128, BIOS @ 0x????????/131072
[     7.519] (II) LoadModule: "glx"
[     7.520] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     7.522] (II) Module glx: vendor="X.Org Foundation"
[     7.522] 	compiled for 1.20.4, module version = 1.0.0
[     7.522] 	ABI class: X.Org Server Extension, version 10.0
[     7.643] (==) Matched nouveau as autoconfigured driver 0
[     7.643] (==) Matched modesetting as autoconfigured driver 1
[     7.643] (==) Matched fbdev as autoconfigured driver 2
[     7.643] (==) Matched vesa as autoconfigured driver 3
[     7.643] (==) Assigned the driver to the xf86ConfigLayout
[     7.643] (II) LoadModule: "nouveau"
[... 2259 lines cut ...]
[  166.647788] pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
[  166.647808] pcieport 0000:00:1c.4: AER: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[  166.647811] pcieport 0000:00:1c.4: AER:   device [8086:a114] error status/mask=00000001/00002000
[  166.647812] pcieport 0000:00:1c.4: AER:    [ 0] RxErr                 
[  166.718506] pcieport 0000:05:02.0: Refused to change power state, currently in D3
[  167.179220] pcieport 0000:05:02.0: Refused to change power state, currently in D3
[  167.179229] xhci_hcd 0000:3a:00.0: Refused to change power state, currently in D3
[  167.257960] xhci_hcd 0000:3a:00.0: Refused to change power state, currently in D3
[  167.257994] xhci_hcd 0000:3a:00.0: Controller not ready at resume -19
[  167.257995] xhci_hcd 0000:3a:00.0: PCI post-resume error -19!
[  167.257996] xhci_hcd 0000:3a:00.0: HC died; cleaning up
[  167.258072] xhci_hcd 0000:3a:00.0: remove, state 4
[  167.258075] usb usb4: USB disconnect, device number 1
[  167.258275] xhci_hcd 0000:3a:00.0: USB bus 4 deregistered
[  167.258278] xhci_hcd 0000:3a:00.0: remove, state 4
[  167.258280] usb usb3: USB disconnect, device number 1
[  167.258465] xhci_hcd 0000:3a:00.0: Host halt failed, -19
[  167.258466] xhci_hcd 0000:3a:00.0: Host not accessible, reset failed.
[  167.258605] xhci_hcd 0000:3a:00.0: USB bus 3 deregistered
[  167.759469] pci_bus 0000:06: busn_res: [bus 06] is released
[  167.759537] pci_bus 0000:07: busn_res: [bus 07-39] is released
[  167.777690] pci_bus 0000:3a: busn_res: [bus 3a] is released
[  167.777773] pci_bus 0000:3b: busn_res: [bus 3b-6e] is released
[  167.777831] pci_bus 0000:05: busn_res: [bus 05-6e] is released
```

EricBoiseLGSVL, 2020-04-08:
@takanotume24 did you find a solution?  We have updated simulator with a new Unity version that fixes many Vulkan issues.  This should be out within the next few days.  If this doesn't help, please re-open this issue

closed 2020-04-08 by EricBoiseLGSVL

lemketron, 2020-04-29:
@takanotume24 The problem may be that the M1000M GPU (Maxwell architecture) is not only old, but appears to only have 2GB of GPU memory:
https://www.techpowerup.com/gpu-specs/quadro-m1000m.c2739

The simulator requires at least 4GB and preferably 8GB. That means using a NVIDIA GTX1070 or 1080 (which are 8GB) or possibly a GTX 1060 (which only has 4GB so it wouldn't be able to load complex environments).

You could also use a GTX 1660 which I believe has 6GB, or a RTX 2070 or 2080 which have 8GB but note that if you’re intending to run Apollo on the same machine, you’ll want a 1070 or 1080 because Apollo 5.0 doesn’t support the newer Turing architecture RTX cards.

In any case it looks like the problem could be related to the video card being too old and (more likely) that 2GB is not enough GPU memory for the Simulator to launch. If you can try a newer more capable GPU you might have better results.

takanotume24, 2020-04-29:
@lemketron Thank you for sharing the details.
Incidentally, the VRAM of M1000M is customized to 4GB though it is limited to the model of my notebook PC.
I haven't tried to fix the problem after I created this issue, but it didn't boot up this time, so it's probably an old GPU problem, not a lack of VRAM.
I'll give it a try when I get a new GPU. Thank you.

EricBoiseLGSVL, 2020-04-29:
@takanotume24 Sorry that we couldn't help more.  Good luck and let us know if you have any issues with the new hardware