lgsvl/simulator #604: Feature discuss:Why not add a trigger to the python NPC "change_lane" behavior?

daohu527, 2020-01-17:
I notice that the python api for controlling npc vehicle has a follow waypoint method, which have a "trigger" property.
Why don't we add a trigger condition to "change_lane", so it will be more convenient to test some scenarios such as "cut in". We don't need to change the position and time of the lane change of npc vehicles, just set some conditions.

shalinmehtalgsvl, 2020-01-17:
Hi @daohu527 ,

Making scenario generation easier is something that we're currently working on.

What you could do right now is the following:
1. Spawn the NPC at the desired location (Point A)
2. Make a list of 2 waypoints at Point A
    - The first waypoint will have a trigger_distance defined for how far the EGO should be before the NPC starts it's lane change
3. Define an on_waypoint_reached() for the NPC
    - In this you can then call follow_closest_lane() and then change_lane()

This should work to get the NPC to change lanes.

Something like this:
```python
sim = lgsvl.Simulator()
point = lgsvl.Vector(x,y,z)
state = lgsvl.AgentState()
state.transform = sim.map_point_on_lane(point)

npc = sim.add_agent("SUV", lgsvl.AgentType.NPC, state)

def on_waypoint_reached(agent, index):
    npc.follow_closest_lane(11.176)
    npc.change_lane(True)

waypoints = []
waypoints.append(lgsvl.DriveWaypoint(point, 11.176, state.rotation, trigger_distance = 30))
waypoints.append(lgsvl.DriveWaypoint(point, 11.176, state.rotation))

npc.follow(waypoints)
```

daohu527, 2020-02-03:
thx! That may be a good solution!

closed 2020-02-03 by daohu527