lgsvl/simulator #605: Perception toggles off automatically after enabling inside dreamview

YilmazOrkun, 2020-01-17:
When I follow these instructions https://github.com/lgsvl/apollo-5.0#launching-apollo-alongside-the-simulator

and enable perception module inside dreamview it toggles off automatically after few seconds.
I could not find anything useful in the log files.

Does anyone have a clue why that happens?

daohu527, 2020-01-17:
The only thing that can be determined is that the prediction module is down. You need to check the apollo log. In addition, you can use "cyber_monitor" to confirm whether the simulator sends a message.

martins-mozeiko, 2020-01-17:
Perception in Apollo usually crashes and does not work if there are not images being published. Check that Simulator is really publishing camera images.

closed 2020-09-25 by lemketron