lgsvl/simulator #606: Lidar topic not visible in cyber_monitor

YilmazOrkun, 2020-01-17:
I would like to use the simulator to collect record files which contains lidar point cloud data.
When I run the simulator I can see that there are messages in /apollo/sensor/lidar128/compensator/PointCloud2 in the bridge view.

But when I launch cyber_monitor the pointcloud topic is not listed there.

daohu527, 2020-01-17:
Do you run the bridge.sh in apollo. You can refer to https://github.com/lgsvl/apollo-5.0#launching-apollo-alongside-the-simulator

You should check the step"Launch bridge (inside docker container):"

YilmazOrkun, 2020-01-17:
I have launched the bridge.
I can see topics from other sensors with cyber_monitor, but not the Pointcloud topic

daohu527, 2020-01-17:
Does your car really have a lidar configuration? If so, you need to check the simulator or apollo to see if there are any errors. By the way do you modify the lidar proto file?

YilmazOrkun, 2020-01-17:
Yes it has. In the simulator when I look to the bridge there is a pointcloud topic which is publishing messages. I can also visualize the pointcloud inside simulator. I did not modify any lidar proto file.

martins-mozeiko, 2020-01-17:
Can you please show screenshot of bridge view in Simulator that shows lidar topic? And also screenshot of cyber_monitor.

closed 2020-09-25 by lemketron