lgsvl/simulator #607: Lidar publishes a point at world origin

Roman-jj, 2020-01-20:
Hello,
I encountered a weird bug with the Lidar.
The PointCloud message published through ROS contains a point right at the world origin. I attached a screenshot showing the point in Rviz.
When moving the vehicle, this point stays at the world origin instead of moving with the vehicle. The compensated option is set to true.
When compensated is set to false, this point moves with the vehicle. All the other points move with the vehicle, regardless of compensated setting.
The intensity value of the point is always 0.
Sensor parameters to reproduce are {"type": "Lidar", "name": "Lidar Velodyne",
            "params": {"LaserCount": 32, "MinDistance": 3.0, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "simulator/lidar_velodyne", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 3.3, "z": 1, "pitch": 0, "yaw": 0, "roll": 0}}

Thanks for your help!

[image]

martins-mozeiko, 2020-01-21:
Could you please record rosbag where this is reproduced, and share with us? Then we'll the details how this bug can happen.

Roman-jj, 2020-01-21:
Hi, sure, I recorded two rosbags. Both are recorded with Jaguar 2015 XE on CubeTown map, spawning at (0, 0, 0), same sensor parameters as above but sensor transform set to (0, 2, 1).
The first one is with compensated set to false, the point moves with the vehicle, always at (0,0,0) in sensor coordinates.
The second rosbag has compensated set to true, the point is not well visible as it stays at world origin while the vehicle moves, but its intensity value is 0, that might help to filter it out.

I already tried looking through LidarSensor.cs but wasn't able to find the cause.

[rosbags.zip](https://github.com/lgsvl/simulator/files/4089996/rosbags.zip)

EricBoiseLGSVL, 2020-04-08:
@rongguodong This issue was fixed correct?

closed 2020-10-20 by EricBoiseLGSVL