lgsvl/simulator #608: click simulator run botton, then it close suddenly.

Fred159, 2020-01-21:
I downloaded lgsvlsimulator-linux64-2019.12 and unzip, then run simulator.

As below figure, I clicked the 'run', it suddenly closed.
[image]

I installed libvulkan1.
my computer is nuvo-7160gc. NVIDIA Graphic card. 8th-gen core intel. i7-8700, 32GB memory, NVMe. Hard disk 1TB. 
I think computer is good enough.

The terminal log information is shown as below.

> aeco_nv@aeco-nv-NUVO:~/lgsvlsimulator-linux64-2019.12$ ./simulator
> Set current directory to /home/aeco_nv/lgsvlsimulator-linux64-2019.12
> Found path: /home/aeco_nv/lgsvlsimulator-linux64-2019.12/simulator
> Mono path[0] = '/home/aeco_nv/lgsvlsimulator-linux64-2019.12/simulator_Data/Managed'
> Mono config path = '/home/aeco_nv/lgsvlsimulator-linux64-2019.12/simulator_Data/MonoBleedingEdge/etc'
> Preloaded 'liblaszip.so'
> Preloaded 'libpng.so'
> Preloaded 'libsqlite3.so'
> Preloaded 'libturbojpeg.so'
> Preloaded 'ScreenSelector.so'
> Display 0 'default': 1920x1080 (primary device).
> Loading player data from /home/aeco_nv/lgsvlsimulator-linux64-2019.12/simulator_Data/data.unity3d
> Logging to /home/aeco_nv/.config/unity3d/Editor/Player.log
> aeco_nv@aeco-nv-NUVO:~/lgsvlsimulator-linux64-2019.12$ 

What is the problem??? I am looking for help~~

martins-mozeiko, 2020-01-21:
It's hard to toll from this what could be wrong.

Please check the log file at `~/.config/unity3d/Editor/Player.log` location. Maybe it has some more useful information at bottom. What kind of GPU do you use? It should be mentioned also somewhere at top of log file.

Fred159, 2020-01-22:
> It's hard to toll from this what could be wrong.
> 
> Please check the log file at `~/.config/unity3d/Editor/Player.log` location. Maybe it has some more useful information at bottom. What kind of GPU do you use? It should be mentioned also somewhere at top of log file.
> 
> Make sure you have nvidia binary driver installed and are not using open-source nouveau driver.

I installed the nvidia driver, it works! Thank you~

closed 2020-01-22 by Fred159