lgsvl/simulator #609: Custom Environment build fails: Could not find file ... _environment_textures

joshwapohlmann, 2020-01-23:
1. Cloned repo and followed instructions as in the documentation.
2. tested and it works (w/ CubeTown)
2. Created custom map using the [documentation](https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map)
3. Made sure to follow the instructions in [Adding Assets](https://www.lgsvlsimulator.com/docs/assets/)
4. Get this error.

```
FileNotFoundException: Could not find file "C:\Users\CoRoLa_1\Documents\lgsvl\Build\AssetBundles\77f6ea09-213d-4afa-a8db-d7495c8f9d5d_environment_textures"
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <7d97106330684add86d080ecf65bfe69>:0)
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
(wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare)
System.IO.File.Open (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share) (at <7d97106330684add86d080ecf65bfe69>:0)
ICSharpCode.SharpZipLib.Zip.StaticDiskDataSource.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate.GetSource () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.AddEntry (ICSharpCode.SharpZipLib.Zip.ZipFile workFile, ICSharpCode.SharpZipLib.Zip.ZipFile+ZipUpdate update) (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.RunUpdates () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
ICSharpCode.SharpZipLib.Zip.ZipFile.CommitUpdate () (at <d353c160b0bc4f95b4c40118f95da2f8>:0)
Simulator.Editor.Build.RunAssetBundleBuild (System.String folder, System.Collections.Generic.List`1[T] environments, System.Collections.Generic.List`1[T] vehicles) (at Assets/Scripts/Editor/Build.cs:447)
Simulator.Editor.Build.OnGUI () (at Assets/Scripts/Editor/Build.cs:264)
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <7d97106330684add86d080ecf65bfe69>:0)
System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) (at <7d97106330684add86d080ecf65bfe69>:0)
UnityEditor.HostView.Invoke (System.String methodName, System.Object obj) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:340)
UnityEditor.HostView.Invoke (System.String methodName) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:334)
UnityEditor.HostView.InvokeOnGUI (UnityEngine.Rect onGUIPosition, UnityEngine.Rect viewRect) (at C:/buildslave/unity/build/Editor/Mono/HostView.cs:310)
UnityEditor.DockArea.DrawView (UnityEngine.Rect viewRect, UnityEngine.Rect dockAreaRect, System.Boolean floatingWindow, System.Boolean isBottomTab) (at C:/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:373)
UnityEditor.DockArea.OldOnGUI () (at C:/buildslave/unity/build/Editor/Mono/GUI/DockArea.cs:340)
UnityEngine.UIElements.IMGUIContainer.DoOnGUI (UnityEngine.Event evt, UnityEngine.Matrix4x4 parentTransform, UnityEngine.Rect clippingRect, System.Boolean isComputingLayout, UnityEngine.Rect layoutSize) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:295)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e, UnityEngine.Matrix4x4 worldTransform, UnityEngine.Rect clippingRect) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:481)
UnityEngine.UIElements.IMGUIContainer.HandleIMGUIEvent (UnityEngine.Event e) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:464)
UnityEngine.UIElements.IMGUIContainer.HandleEvent (UnityEngine.UIElements.EventBase evt) (at C:/buildslave/unity/build/Modules/UIElements/IMGUIContainer.cs:444)
UnityEngine.UIElements.MouseCaptureDispatchingStrategy.DispatchEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/Events/MouseCaptureDispatchingStrategy.cs:93)
UnityEngine.UIElements.EventDispatcher.ProcessEvent (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:280)
UnityEngine.UIElements.EventDispatcher.Dispatch (UnityEngine.UIElements.EventBase evt, UnityEngine.UIElements.IPanel panel, UnityEngine.UIElements.DispatchMode dispatchMode) (at C:/buildslave/unity/build/Modules/UIElements/EventDispatcher.cs:156)
UnityEngine.UIElements.BaseVisualElementPanel.SendEvent (UnityEngine.UIElements.EventBase e, UnityEngine.UIElements.DispatchMode dispatchMode) (at C:/buildslave/unity/build/Modules/UIElements/Panel.cs:189)
UnityEngine.UIElements.UIElementsUtility.DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel panel) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:255)
UnityEngine.UIElements.UIElementsUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/UIElements/UIElementsUtility.cs:78)
UnityEngine.GUIUtility.ProcessEvent (System.Int32 instanceID, System.IntPtr nativeEventPtr) (at C:/buildslave/unity/build/Modules/IMGUI/GUIUtility.cs:179)
```

Thing is, I tried the SingleLane example as well. Is practically the same.
* I have map origin, i have spawn, i have map annotations.
* There is no other error.

mark-gerow-lge, 2020-01-23:
Hey Josh,

That error seems to indicate that the map is failing to build the texture bundle for the given map. When we build maps/vehicles we build three Unity AssetBundles into our bundles - windows specific assets, linux specific assets and textures. If your map uses no textures / the textures used are not in your external/map folder, it will not build the texture bundle and throw an error when it tries to add it to the finished map bundle. We are introducing a fix to this for the case of users who want to make texture-less maps which should be in the next release.

If your custom map folder does have all the requisite textures, could I have you break point debug building your custom map on line 407 of Build.cs: 

`var textureBuild = new AssetBundleBuild()
                    {
                        assetBundleName = $"{manifest.bundleGuid}_environment_textures",
                        assetNames = AssetDatabase.GetDependencies(sceneAsset).Where(a => a.EndsWith(".png") || a.EndsWith(".jpg")).ToArray(),
                    };`

and step through and see what is in textureBuild.assetNames?

-Mark

joshwapohlmann, 2020-01-24:
Hey Mark,

I didn't used ANY textures. Thanks for pointing that out. After using at least 1 texture in my environment, the build was successfull.

closed 2020-01-26 by joshwapohlmann