lgsvl/simulator #610: How to give each pedestrian a different color in Semantic Camera?

RamoramaInteractive, 2020-01-24:
According to https://www.lgsvlsimulator.com/docs/sensor-visualizers/#semantic-camera pedastrians have the semantic color red. Is it possible to set up the semantic color to give each single pedestrian a different color value?

martins-mozeiko, 2020-01-24:
Current code does not allow this.
But there is pull request that implemented such functionality: https://github.com/lgsvl/simulator/pull/453
Unfortunately it gives significant performance penalty. You should check if that is acceptable in your use case.

RamoramaInteractive, 2020-01-28:
I changed the Script files but after starting the new compiled Simulator I received following error:

```
Traceback (most recent call last):
  File "run_sim.py", line 23, in <module>
    a = sim.add_agent("Lincoln2017MKZ (Apollo 5.0)", lgsvl.AgentType.EGO, state)
  File "c:\users\myComputer\desktop\pythonapi\lgsvl\utils.py", line 21, in new_f
    return f(*args, **kwargs)
  File "c:\users\myComputer\desktop\pythonapi\lgsvl\simulator.py", line 110, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "c:\users\myComputer\desktop\pythonapi\lgsvl\remote.py", line 68, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object
```

zelenkovsky, 2020-01-31:
Well, that's difficult to answer if we do not know what did you change in the script.

Do you still have "Lincoln2017MKZ (Apollo 5.0)" vehicle?

RamoramaInteractive, 2020-01-31:
@zelenkovsky I've included all the commits from https://github.com/lgsvl/simulator/pull/453 to its appropiate cs-files and also included the missing files into my Unity simulator.

Yes, I'm still using the "Lincoln2017MKZ (Apollo 5.0)" vehicle.

RamoramaInteractive, 2020-03-07:
After 2 months how can I find out if I included the right source Codes from https://github.com/lgsvl/simulator/pull/453 into my Unity Simulator?

Can I still use the Semantic Camera after enabling the Semantic Segmentation?

EricBoiseLGSVL, 2020-03-09:
@RamoramaInteractive Yes, we have looked at the merge request and it should be fine.  We have implemented the same feature and will be releasing it soon.

EricBoiseLGSVL, 2020-04-08:
@RamoramaInteractive we are releasing the instance semantic as a JSON setting in semantic camera sensors.  Thank you for your contribution.  The next public release should be in a few days

closed 2020-04-08 by EricBoiseLGSVL