lgsvl/simulator #611: Increase the road length of SingleLaneRoad map

junjieshen, 2020-01-24:
Hi,

Sorry if I post the question in the wrong place since I couldn't file an issue in the SingleLaneRoad repo. 

I was wondering if there is any way to increase the road length of the SingleLaneRoad map. Currently, it seems the length is around 200m. I would like to increase the length for testing the driving behavior in a long duration.

Thanks,
Junjie

EricBoiseLGSVL, 2020-01-27:
@junjieshen 

We are looking to create a longer test map but we are not sure when we will be able to release it.  In the meantime, you can open SingleLaneRoad in Editor, duplicate the road mesh as many times as you need and build a new map.

junjieshen, 2020-01-28:
@EricBoiseLGSVL 

Thank you for the suggestion. Duplicating the mesh did the trick for me!

closed 2020-01-28 by junjieshen