lgsvl/simulator #612: Using three camera sensors at the same time

RamoramaInteractive, 2020-01-25:
I want to capture an image and save it as Color image, semantic image and depth image.
Is it possible to declare three sensor objects of the sensor object and capture an image at the same time?

mark-gerow-lge, 2020-01-27:
I am unsure of the exact question here, you can place a Color, Depth, and Semantic sensor on your vehicle at the same time.

shalinmehtalgsvl, 2020-01-27:
Hi @RamoramaInteractive ,

You can use the Python API to save images from the cameras or you can subscribe to the ROS topics published by the sensors.

You would need to add all 3 cameras to a vehicle and make sure the parameters are all the same for you to get the same perspective in the images.

closed 2020-04-08 by EricBoiseLGSVL