lgsvl/simulator #613: follow autoware-instructions and not working

yugushihuang, 2020-01-27:
Hello,

I just want to replicate what in the autoware instruction. 

https://www.lgsvlsimulator.com/docs/autoware-instructions/

I got several problem:
(1) I use NVIDIA docker container as stated in the instruction. First, as it states Click "Map" to load the launch file pertaining to the HD maps. An "Ok" should appear to the right of the "Ref" button when successfully loaded.  When I click "Map" in my autoware, there is no "OK" button appear. Do you know what the problem might be?
(2) Why use 9090 to connect to autoware? Is that specify in somewhere? Apparently, my LGSVL simulator is not connect to autoware.

Thanks!

EricBoiseLGSVL, 2020-01-29:
@yugushihuang 

Let me ping @shalinmehtalgsvl that may be able to help

shalinmehtalgsvl, 2020-01-29:
Hi @yugushihuang ,

1) "Ok" appears for me when running BorregasAve, but not always with the rest of the maps. If you open RViz is the point cloud shown? 

2) 9090 is specified somewhere in Autoware. You need to enable `Sensing` which will enable the ROS bridge. Then the simulator will connect to Autoware.

yugushihuang, 2020-02-03:
@shalinmehtalgsvl 
Thanks for the reply!

If I open RViz, the point cloud is not showing.

hadiTab, 2020-02-04:
@yugushihuang can you check the status of your bridge?
To do that, once you have everything running, click on the menu button at the bottom left of the screen and then on the bridge button:
[image]

Does the 'Bridge Status' say 'Connected'?

yugushihuang, 2020-02-04:
@hadiTab 

I have checked, it shows "Disconnected"

yugushihuang, 2020-02-04:
@hadiTab

I have checked, it shows "Connected"

I think my problem is autoware cannot load pcd files

hadiTab, 2020-02-04:
@yugushihuang check the terminals that open when autoware is being loaded. As soon as you load the maps, some messages will be logged. If there is an issue loading the files it should print an error message.

yugushihuang, 2020-02-05:
@hadiTab I found out I did not install git lfs before pulling the "autoware-data". Now my problem becoming no time to display to the right of "Ref" after click "Localization".

yugushihuang, 2020-02-05:
@hadiTab 
Error message is ->

terminate called after throwing an instance of 'image_transport::TransportLoadException'
what():  Unable to load plugin for transport 'compressed', error string:
According to the loaded plugin descriptions the class image_transport/compressed_sub with base class type image_transport::SubscriberPlugin does not exist. Declared types are  image_transport/raw_sub

hadiTab, 2020-02-05:
That error message can be ignored for now (it won't affect the localization).
NDT matching localization is a bit finicky. At the very least you will always need to wait an undetermined amount of time for it to find a good match. It's usually helpful to correct the pose manually in rviz using the "2D Pose Estimate" button at the top and drawing an arrow at the right spot pointing in the right direction (generally the vehicle should spawn at the bottom of the map). Driving slowly also tends to help. Regardless, you need to wait a while for it to find the match.

pedroexenberger, 2020-02-07:
> @yugushihuang can you check the status of your bridge?
> To do that, once you have everything running, click on the menu button at the bottom left of the screen and then on the bridge button:
> [image]
> 
> Does the 'Bridge Status' say 'Connected'?

I'm also trying to reproduce the lgsvl + autoware but I got "Bridge Status Disconnected" in the lgsvl simulator. I'm running both autoware and lgsvl in the same computer, and hence defined the connection string as "localhost:9090" when creating the simulation in LGSVL. I've selected and enabled the sensing launch file before starting the simulator, but still cannot make the connection works. Anyone had experienced the same? I'm running on ubuntu 16, autoware 1.12.0, and lgsvl 2020.01-rc1 (just downloaded).

hadiTab, 2020-02-08:
@pedroexenberger can you check the autoware terminal when you launch the sensing launch file and see if there are any errors related to the bridge?

pedroexenberger, 2020-02-10:
@hadiTab When I launch the sensing file, the following appears on the autoware terminal:

```
['roslaunch', '/home/autoware/shared_dir/autoware-data/BorregasAve/my_launch/my_sensing_simulator.launch']
pid=1973
sched policy=OTHER prio=0
terminate called after throwing an instance of 'image_transport::TransportLoadException'
what():  Unable to load plugin for transport 'compressed', error string:
According to the loaded plugin descriptions the class image_transport/compressed_sub with base class type image_transport::SubscriberPlugin does not exist. Declared types are  image_transport/raw_sub
[republish-2] process has died [pid 1994, exit code -6, cmd /opt/ros/kinetic/lib/image_transport/republish compressed in:=/simulator/camera_node/image raw out:=/image_raw __name:=republish __log:=/home/autoware/.ros/log/bab32062-4be1-11ea-a0b6-704d7b89295f/republish-2.log].
log file: /home/autoware/.ros/log/bab32062-4be1-11ea-a0b6-704d7b89295f/republish-2*.log
```

I'm not very familiar with the project yet, so not really sure of what this error means. Is there a problem with 'image_transport' that is making the communication to terminate? Also, there is no republish-2.log file in the log folder to paste information here.

pedroexenberger, 2020-02-10:
So for some reason, the autoware docker does not contain the package `ros-${ROS_DISTRO}-image-transport-plugins`. Installing it with `apt-get update` and `apt-get install ros-${ROS_DISTRO}-image-transport-plugins` inside the autoware docker solved the error message when launching the sensing file. 

However, the bridge status remains as 'disconnected' inside the LGSVL simulator.

pedroexenberger, 2020-02-10:
Tried in another machine (Ubuntu 18.04, ROS melodic, and autoware 1.13.0) and the connection worked (appears connected in the simulator). Is there any other place where I could find a log or some clue to understand why the bridge does not work within my environment (Ubuntu 16.04, ROS kinetic, autoware 1.12.0)? I tried to roslaunch only the bridge (as in the my_sensing_simulator.launch), but no errors are reported there also. Thanks!

pedroexenberger, 2020-02-10:
OK, solved. 

Instead of `localhost:9090` in the connection string, in the LGSVL simulation configuration, I have set  `computer-name:9090`.  Not sure why I needed to do that, but the connection works now :)

yugushihuang, 2020-02-17:
@hadiTab Sorry for the late reponse, I still cannot get the example running. 
(1)  I had the error message of  "failed createPoleMarker" when I load the point cloud map in autoware;
(2) How can I draw the arrow in RViz? I did not find the option in RViz to do that.
(3) Failed to find match for field 'intensity' after clicking localization.

Thanks!

yugushihuang, 2020-02-17:
@hadiTab 
I have checked on autoware side, the vehicle model has error: vehicle model parameter robot description does not exist.

yugushihuang, 2020-02-17:
@pedroexenberger Good to hear you solve the problem! I still cannot duplicate the example. There is no vehicle show up in my RVIz!

yugushihuang, 2020-02-17:
@hadiTab 
[image]

pedroexenberger, 2020-02-17:
Hi @yugushihuang.
Some hints from my recent experience.

> @hadiTab Sorry for the late reponse, I still cannot get the example running.
> (1) I had the error message of "failed createPoleMarker" when I load the point cloud map in autoware;
> (2) How can I draw [...]

(1) and (3) are not actually a problem to reproduce the start guide. The same happened to me and I could do it even though. (1) is a problem that happens while loading the map on rviz (not sure why exactly), but has been mentioned in other issues (e.g., https://github.com/lgsvl/simulator/issues/411#issuecomment-538521227). The solution for some people is to checkout a previous commit of autoware-data; but as I told you, I could reproduce the start guide even with those error messages. About (3), from what I understood reading other issues (e.g., https://github.com/lgsvl/simulator/issues/51#issuecomment-448521053), autoware expects an 'intensity' field while parsing/reading the .pcd (point cloud) files (hd map), but they are not provided in the autoware-data maps. Again, that is not what is preventing you to reproduce the start guide as it also occured to me. 

As for (2), I think @hadiTab was referring to the '2D pose estimate' button, in the upper part of the RViz window. Basically you click there, and then click on the point of the road your car is supposed to be, and drag the mouse towards the direction the car is facing. There is an example of using the '2D pose estimate' functionality [here](https://youtu.be/VnJxeKnvF5s?t=56). This step is important to reproduce the start guide, and you should try to place the car as better as possible based on the example images on the start guide. This helps the ndt algorithm to localize the car in the map.

> @hadiTab
> [image]

Can you launch my_mission_planning.launch file after having the LGSVL and RViz set up? When you do that, the possible lanes to drive should appear in RViz, like in [this image](https://www.lgsvlsimulator.com/docs/images/autoware-ndt-matching.png), where you can see the red/orange bold lines. When you start LGSVL, the car is positioned along one of those lines (in the very beginning of it, as you can see in the image). You should use the '2D pose estimation' to click at the beginning of the lane, dragging the mouse along the direction of the lane.

If that doesn't work, check the output from the autoware terminal whenever you launch a .launch in autoware, and check if any error is reported. I'm not sure of that, but apparently, when one of the 'instructions' in a .launch file fails, the ongoing launching is 'aborted', and some nodes that should be initiated on that .launch file might not start because of a previous error when launching another node. 

Hope it helps you!

yugushihuang, 2020-02-18:
@pedroexenberger Thank you so so much for such a thorough and detailed answer! 
I have never tried launch my_mission_planning.launch as I strictly follow the instruction of first clicking map then sensing. And supposedly, the vehicle should appear. But in my case, the vehicle never appear and I got vehicle model error message in Rviz as the pic I attached. I have never saw the red line appear at all. Is the 2D estimate pose button usable only when the red line appears?

Thank you again!

pedroexenberger, 2020-02-18:
@yugushihuang you are welcome. 

> And supposedly, the vehicle should appear

Do you mean a draw of a vehicle in RViz? You actually don't need it. What is important is to have the sensors working. That will be clear when the lidar point cloud appears in RViz. I just ignored the error in the vehicle module. I think the problem is that you are asking RViz for a topic that is not provided by anyone. I think you can uncomment the `` line inside the `my_localization.launch` file. But then you will also have to adjust the `vehicle_model.launch` file properly (you have to find it within the install folder). I don't think it is worthy to invest time in that because you don't need it for the start guide to succeed.

> Is the 2D estimate pose button usable only when the red line appears?

I don't think so. It is just more convenient to have them displayed so you can find the middle of the road when you place the car.

> I have never saw the red line appear at all.

Those will appear only after you run my_mission_planning.launch (and having the appropriate display variable in RViz enabled).

yugushihuang, 2020-02-18:
@pedroexenberger Thank you very much!

So for the start guide to succeed, I should get my vehicle in LGSVL move autonomously(driving by autoware) right? I think I have the point cloud correctly loaded. But still no, after I click the run button in LGSVL, my vehicle just stay still.

I have viewed the youtube video you attached of using the '2D pose estimate' functionality. I cannot find an open planner like the video. And if I right click 2D pose estimate, there is a green arrow, but if I let go the right click, the green arrow just disappear too. 

I did not get the part of bright red concentric circles around the vehicle. Is that mean my sensing is not working?

[image]

Sorry I have so many questions.
Thank you very much!

pedroexenberger, 2020-02-18:
@yugushihuang The video was just to show you how to use the 2D pose estimation (placing the arrow in the road, that's it), all the rest is in a different context, so don't take it in account.

Also, you have to play the LGSVL simulation (the play button) before you use the 2D pose estimation. The LGSVL simulation is what provides the lidar data to autoware (and which can be visualized as the concentric red dots in RViz). So you have to launch my_map, my_sensing, my_localization, (optionally you can also launch my_detection and my_mission_planning), and play the LGSVL simulation. Only then you proceed with the 2D pose estimation step. Once your pose estimation is sufficiently good (for the ndt algorithm), the car will find itself in the map, and the lidar data should appear, as explained in the instructions. Also, you should make sure that the LGSVL simulator and autoware bridge are connected, as explained above in this issue https://github.com/lgsvl/simulator/issues/613#issuecomment-582047527.

Hope it helps you.

yugushihuang, 2020-02-21:
@pedroexenberger Sorry for the late response, but it seems my car finally find itself in the map! Thank you so much for all the patience!!!

pedroexenberger, 2020-02-21:
@yugushihuang nice to hear that! I think if that is all you can close the issue :)

closed 2020-02-26 by yugushihuang

xiewenjing1170, 2021-11-10:
@yugushihuang Hello, I meet the similiar problem. I follow the ROSBAG demo (docker Autoware 1.14 + ubuntu 20.04). When I load "my_map.launch" and click "Map" in the quick start page in Autoware. There is no "Ok" to the right of Ref button, and no point cloud in RVIz. There is no any error information in the terminal. Do you know what the problem might be? Thanks.