lgsvl/simulator #614: Support for Apollo 5.5

pedroexenberger, 2020-01-29:
Hi all, 
Are there many differences between lgsvl Apollo fork comparing to the baidu's master repo? I was just considering to setup the 'official' apollo 5.5 version bridge and try it out with the lgsvl simulator.
Is that not recommended? 

I haven't started anything yet, so I'd just like to know the best approach (both for correctness, but also as up-to-date as possible). 
Thanks.

zhouyapengzi, 2020-05-02:
Have you successfully bridge 5.5 with lgsvl？ I want to do exactly the same thing.

pedroexenberger, 2020-05-02:
Hi @zhouyapengzi, sorry but I gave up using Apollo and switched to autoware for my purposes. 

I hope someone else can help you on this and, for this reason, I'm reopening the issue.

rongguodong, 2020-05-03:
We are actively working on supporting latest Apollo. Will post updates here soon.

lemketron, 2020-05-12:
> We are actively working on supporting latest Apollo. Will post updates here soon.

[Several PRs](https://github.com/ApolloAuto/apollo/pulls?q=is%3Apr+author%3Arongguodong+%22LGSVL+Simulator%22) have been submitted to the ApolloAuto project to add support for LGSVL simulator, several of which have already been accepted. If anyone is eager to test this, you can check out Apollo master and and then manually add in the changes in the unmerged PRs.

We will provide updated documentation for using LGSVL Simulator with Apollo (master) soon.

rongguodong, 2020-05-18:
We have all five PRs merged into Apollo now. The latest Apollo master should works with our simulator. 
@pedroexenberger @zhouyapengzi You can clone latest Apollo and test it. You can follow most of instructions to Apollo 5.0 with two changes:
- Call "bootstrap_lgsvl.sh" instead of "bootstrap.sh" to launch Dreamview; and
- Select  the `Mkz Lgsvl` mode, in addition to select `Lincoln2017MKZ LGSVL` vehicle and `BorregasAve` map.
Please let us know if you found any problems when using latest Apollo with LGSVL Simulator.

ntutangyun, 2020-05-31:
@rongguodong thanks for the PRs submitted to the Apollo 5.5 master. 

I noticed that currently, only the Borregas Ave is available. I guess I could manually add the maps in. 

Besides the maps, I wonder if the Apollo 5.5 master branch is at a similar level of integration (between LGSVL and Apollo) compared to forked 5.0 version. 

I mean if there are any missing functions that's implemented in the forked integration solution than the master branch. 

Thanks

rongguodong, 2020-06-01:
Yes, for other maps, the temporary workaround is to copy the one you need from "https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data". 

For vehicles, Lincoln2017MKZ is currently added into Apollo master (renamed to "Lincoln2017MKZ_LGSVL"), and you can copy others from "https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/calibration/data".

Besides some maps and vehicles missing, latest Apollo master should work the same as our fork of Apollo 5.0. If you find anything wrong, please let me know.

ntutangyun, 2020-06-02:
@rongguodong great. thanks.

lemketron, 2020-09-25:
> seems that they are using different algorithm configs defined here.

Yes, the Mkz LGSVL configuration uses RTK localization rather than MSF (and a different value for lidar_query_tf_official in the LiDAR perception configuration file).

lemketron, 2020-09-25:
Instructions for using LGSVL Simulator with Apollo master (~6.0) can be found here: https://www.lgsvlsimulator.com/docs/apollo-master-instructions/

closed 2020-09-25 by lemketron